jnsbyr/esp8266-intexsbh20 #2: Need more information

PaTiTan, 2021-05-31:
Hello,
You are doing a great job.

Can you give us the list of components?
Also general diagram of all the architecture because I do not know if your module is in parallel or in series.
Thanks

B4n4n3, 2021-05-31:
Hello jnsbyr,

thanks for your great job!
Can you post the links to the german stores for the components?
i am interessted in 

- IP64 case
- connector and cables

thanks a lot!

jnsbyr, 2021-05-31:
Here is a list of the components that are not obvious from the schematic:

| description | part no. | URL |
| - | - | - |
| main case | BOPLA ET-205 | https://www.reichelt.de/de/en/housing-52x50x37-mm-bopla-et-205-p5725.html
| y splitter case | RX2003-S-5 | https://www.reichelt.de/de/en/kunststoffgehaeuse-26-x-18-x-13-mm-5-stueck-rx2003-s-5-p292742.html
| 5pin M12 plug with cable | LUT 1210 05 06 | https://www.reichelt.de/de/en/sensor-cable-m12-5-pin-male-female-0-6-m-lut-1210-05-06-p223002.html
| PG cable gland | PGBF 7 | https://www.reichelt.de/de/en/cable-gland-pg-7-3-6-5-mm-pgbf-7-p126186.html
| PG lock nut | PGGM 7 | https://www.reichelt.de/de/en/locknut-for-pgbf7-pggm-7-p126192.html?search=PGGM+7&&r=1
| 10k NTC | EPC B57164-K103 | https://www.reichelt.de/de/en/ntc-resistor-450-mw-10-kohm-epc-b57164-k103-p240041.html
| fuse | ESKA 883.110 | https://www.reichelt.de/de/en/radial-fuse-time-lag-t-200ma-square-eska-883-110-p279212.html

jnsbyr, 2021-05-31:
@PaTiTan
> Also general diagram of all the architecture because I do not know if your module is in parallel or in series.

Please have a look at the schematic. It shows a block named "Control Panel Bus Tap". It should imply that the circuit is connected in parallel. In the description above this is mentioned explicitly "the original control panel and an additional remote control unit can be attached in parallel to the mainboard because the signalling supports a multi-drop architecture for listening".

You may also have look at the top left of the [schematic from Geoffrey](https://github.com/YorffoeG/diyscip/blob/master/docs/schematic_PCB_V2.jpg). 

Hope this helps.

jnsbyr, 2021-05-31:
Application note for the assembly of the y splitter case:

The part mentioned in the table above is tiny in comparison to the dimensions of the cable. The drill holes I used are a little smaller than the cable diameter to make them self sealing. Originally I planned to fill the case with epoxy to prevent the cables from being pulled out accidentality and to make it waterproof. But the fit was so tight that it was not really necessary. The cable ends need to be soldered together. To prevent a short circuit the open ends must be isolated again, e.g. using shrink tubing.

If you don't intend to spend more than an hour on this item, use a case with a volume at least 3 time larger instead. You might even skip the soldering if you use a ready to use connector with screws or clips.

B4n4n3, 2021-06-01:
Thanks a lot for the partlist. Did you use the original cable (mainboard <-> controle panel)?

jnsbyr, 2021-06-01:
Yes, please have look at the photo of the [image]. Looking from the position of the y spitter case: 
- the "center" cable to the control panel is about 2/3 of the original cable from the mainboard to the control panel
- the "right" cable with the Intex plug is about 1/3 of the original cable from the mainboard to the control panel
- the "left" cable to the WiFi controller is the M12 cable from the component list, approximately split in half

PaTiTan, 2021-06-05:
[image]
but what is this grey thing?

jnsbyr, 2021-06-06:
You might have guessed by elimination that this must be the NTC. You can
also use the link in the component table for another image of the same
component. NTCs come in various makes and sizes. I chose this because it
was cheap and comparatively robust (large). Any other 10k NTC will do.

PaTiTan, 2021-06-08:
last question : where is number 1 pin ? 
Left or right connector : 
[image]

PaTiTan, 2021-06-08:
The "command" dont appear so do it myself but i dont understant how work this command : 

`pool/command/water/tempSet | int | °C`

i just want to increase or decrease or put a value of température wanted 
I need something too is what température i have ordered. (pool/water/tempAct is what temperature is inside).

thanks for your incedible job !!

jnsbyr, 2021-06-08:
> The "command" dont appear ...
This is an aspect of your MQTT server. Typically you need to define the MQTT topics that are to be send from the server to the client on the server side and these kind of topics typically contain the label "command". So here you need to enter all entries listed under "Subscribed Topics" in the readme.

Once you have established a connection between the spa WiFi controller and your MQTT server you should be able to set any of the "Subscirbed Topics" on the server side. You can use serial debugging to check if the ESP8266 is receiving the MQTT command data.

> I need something too is to see what température i have ordered. ...
With the topic "pool/water/tempAct" you have made the right choice. The spa WiFi controller will send it after the connection is established to the MQTT server and to the spa mainboard.

jnsbyr, 2021-06-08:
> @PaTiTan Where is number 1 pin? https://user-images.githubusercontent.com/69749534/121177003-2dbff280-c85d-11eb-8466-3abf5e1a9a3a.png

In the schematic there is an abstract drawing of the plug titled "PureSpa Plug (M)" in the bottom right corner. The "M" stands for "male" and such a plug should have pins. The schematic shows the plug from its front side. So, looking at your images of the two plugs, the left one would be the M-type (with the contact pins missing) while the right one would be F-type. The socket in the Intex pump housing is the F-Type while the Intex control panel cable has a plug with the M-Type.

PaTiTan, 2021-06-09:
maybe use this from diyscip ;)
[image] 

anyway, THANK YOU SO MUCH for you awesome great work ! 
is working well (exept for ordering temperature with Jeedom via jMQTT).

jnsbyr, 2021-06-09:
> @PaTiTan ... exept for ordering temperature with Jeedom via jMQTT

Thank you for your feedback, I am glad that you like this project.

The drawing of the plug from Geoffroy is good, but I didn't want to copy it
and I am not much of an artist myself.

You say that setting the temperature is not working for you. That's sad
because that was one of the major reasons I started this project. I cannot
help you with anything that is Jeedom specific, but if you are interested
in making this work I will try to help. First you must establish if the
MQTT command for the setpoint temperature reaches the WiFi controller.
Connect TX+GND of the ESP8266 to an TTL/USB converter connected to your PC
and open a serial terminal (e.g. PuTTY) with the serial parameters
74880/8/N/1. Every time you send an MQTT command from Jeedom you should see
a serial output looking like "set: <topic>: <value>". If not, the problem
is probably on the Jeedom side. If the topic "pool/command/water/tempSet"
is received and the value looks OK then contact me again.The value must be
an integer number, it should not contain a decimal point.

PaTiTan, 2021-06-10:
Tanks a lot, it was Jeedom MQTT :)
All work really good. 

Thank you so muck, you are the best!!!

JonOLinux, 2021-06-15:
Hello jnsbyr
Thank you for your great work and for sharing.
I have a SBH-20 and I would like to domotize it, do you sell the card please ?
thank you

jnsbyr, 2021-06-15:
@JonOLinux 
Sorry, I cannot help you in this respect. Please read the Readme.md again. This project was started primarily as a proof of concept. I have not designed a PCB but used a breadboard for the prototype. It is working for me and there is no necessity for me to replace the breadboard with a PCB.

You could use a tool like KiCad and design your own PCB - but really, you don't need it. The circuit has only a few component and you need to solder anyway, so adding a few wires on the backside of the perfboard should be no big deal. It only takes a little longer than using a breadboard.

JonOLinux, 2021-06-15:
Ok, I will do it.
Thank you for your help.

jnsbyr, 2021-06-16:
@JonOLinux 
If you decide to design a PCB for this project, please consider adding it to this repository (e.g. as a merge request). It will certainly make someone else happy.

stevengoossensB, 2021-07-06:
Hi @jnsbyr 

Could you share a picture of the backside of the perfboard and how it's soldered together as a reference? I'm trying to understand how to connect all components in the picture (https://raw.githubusercontent.com/jnsbyr/esp8266-intexsbh20/master/assets/Perfboard.jpg) in a clean manner without unwanted interconnections.

jnsbyr, 2021-07-07:
Hope the picture I was able to make will help. The left is still left but the bottom of the top side picture is the top of the bottom side picture. A 3D picture of the perfboard wiring would be better suited, but alas, that's not done with a click.

There are a total of 4 isolated thread wires for crossovers, all other connections are without crossovers or use pad to pad soldering. I tried to make use of all the wires that come with new components. I a few cases it was enough to just bend them first and cut them off at the next component pad.

[image]

And here is a picture of the open case that shows how the M12 cable connects to the board:

[image]

stevengoossensB, 2021-07-09:
Many thanks, this is really helpful!

closed 2021-07-26 by PaTiTan

Matze5593, 2023-05-11:
Hello,

could someone tell me more about the tendal capacitor?

Tantalum capacitor, 10 µF, 16 V, RM 2.5

What voltage should the capacitor have?

https://www.reichelt.de/tantal-kondensator-10-f-16-v-rm-2-5-tantal-10-16-p20332.html

jnsbyr, 2023-05-11:
> What voltage should the capacitor have?

Looking at the schematic, the cap is connected to +5V nominal, might be around +5.25V for real.
Looking at the datasheet of the linked cap the +10V version is rated for a surge voltage up to +8V. This would be the minimum recommeded value. Any higher value will also work.