jnsbyr/esp8266-intexsbh20 #4: Need some code explantion

titolini72, 2021-11-24:
Hi Jens,
I drop you a request after someone told me about your work. Really nice actually !! 
I already start the same project but from Jof. I follow the schematics from his GitHub but today it doesn't work
- MQTT is ok (verified) : Rx and Tx
- Decoding frames are ok as well 
- Nonetheless button press simulation are not ok at all; and honestly I don't understand the schematics base on SN74HC595 and CD4051.
- 
You solution has the main advantage to perform everything from ESP8266 in sw.

However reading your code I don't understand how it works though :) The main question is how you emulate the "Press Button"
Starting from `poolIO.setPowerOn` (MTTQ Cmd) for example 
`pressButton(buttons.togglePower); `(Still ok) 

How this simple statement ends up to a "Power on button press" ?!?
I see `ICACHE_RAM_ATTR inline void SBH20IO::decodeButton()` but here I don't get how this works either
If Power On is for instance Off, I assume `else if (isrState.frameValue & FRAME_BUTTON::POWER)` is false. ... so my interrogation

Would you mind to clarify this point please

Thanks in advance and best regards

jnsbyr, 2021-11-25:
Hi titolini72,

I will try to help you. I didn't get how Jof's hardware works, either, and he didn't answer my questions.

The whole process of signalling a button action is really time critical. The only way to handle this asynchronously with a MCU is using an ISR. 

So the method `pressButton()` itself does not do much except setting the control value `buttonPressCount` and monitors if the control value changes. `buttonPressCount` is the parameter of `pressButton()` but is not a value parameter but a reference parameter. This results in modifying the values of the `Buttons` struct. Look into the ISR method `SBH20IO::decodeButton()`. Here the `Buttons` are handled by changing `isrState.reply`. And at the end of the method `PIN::DATA` is pulsed.

Hope this helps.

titolini72, 2021-11-26:
> he didn't answer my questions.
Well I had told with him by phone w ... after the call he never replied to any inquiries : SMS, Phone ,e-mail ... doh
> The only way to handle this asynchronously with a MCU is using an ISR.
of course 

Given you explanation, this is my understanding of the code ... so my misunderstanding on how the code should behave.
Let me give you an example
Let say I want to Power on the SPA via MQTT command (`setPowerOn` function) . This commend is supposed to set bit 0x0400(bit 10) in the 16bits frame (Button Power on). 
`SBH20IO::decodeButton()` is called at the very end of the frame, I mean after receiving the 16 bits of the frame. 
**Where this bit 10 is pulled or toggled ?**
Assuming the SPA is OFF, the flag FRAME_BUTTON::POWER is obviously at 0, the reply is obviously at false ... so my confusion :)
` And at the end of the method PIN::DATA is pulsed.` Again confusion :) Pull Up ? Down ? by setting DATA as output create a Pull Down ? BS170 converts this value in 0V TTL ? 

I am very sorry for having so many questions. I hope you are not too upset by such inquiries. 
Best regards

jnsbyr, 2021-11-26:
> Let say I want to Power on the SPA via MQTT command (`setPowerOn` function) . This commend is supposed to set bit 0x0400(bit 10) in the 16bits frame (Button Power on).

No, you are looking at this project from the perspective of Jof's project and Jof never claimed that his project is suitable for the SBH20. Check my [screenshot form the digiscope](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/assets/Button-Signalling.png) and look for the pulse on the data line that is manually marked in red. The pulse comes **after** the data that asks "Is butten XXX pressed?". If the pulse is send after the data (during the clock break) with the right timing, it is accepted as a "yes".

The BS170 works as a bidirectional level shifter. Most of the time the ESP8266 only receives bits (high impedance input). When a pulse must be sent back the data output is temporarily activated and set to low and this also makes the data line go low.

titolini72, 2021-11-27:
`No, you are looking at this project from the perspective of Jof's project `
Yup it looks like. I thought it was the same though. 
`The pulse comes after the data that asks "Is butten XXX pressed?".`
I will put my scope as well. it looks like the button are scanned by the server. The pull is just to confirm the scan. Let me see that ;)
`The BS170 works as a bidirectional level shifter. `
Gotcha I have a BSS138 at my end that works for this same purpose. The main difference is this is surface mounted devices.  Nothing else.

As a conclusion, I believe the design from Jof can be used (I mean I can "reuse" the PCB I have by no using a part of it. On ly Sw has to be changed). I will investigate and let you know

BTW thanks for all these valuable inputs 

Regards

jnsbyr, 2021-11-27:
Glad I could help and I agree that it should be possible to make use of the PCB from Jof for this project by not using the the CD4051BE and SN74HC595N components, but it might be necessary to interrupt the connection of the DATA/CLK/HOLD-TTL signals to these components to avoid side effects.

titolini72, 2022-01-10:
Hi Jens,
Mixing Jof's code and yours I end up in a working solution. The PCB 3/4cm is on the way to finalize.
If I keep Jof's code, this is because how CFG is managed (Web interface to setup Wifi and EEPROM to store data). However the Control Panel is all yours ! I tried to upgrade Jof's but this was a disaster. Copy/Paste + adaptation was the best option 

Thanks for your help we can close the issue ;)

closed 2022-05-08 by jnsbyr