jnsbyr/esp8266-intexsbh20 #5: How to use a different mqtt Port

serb91, 2022-03-23:
Hi,

i am trying to use your Project on a Intex SimpleSpa
At first step i want to flash my d1 mini and also get it in iobroker. 
my problem is, that i have to use a different port for mqtt, for example 4321. 
In you project mqtt is using the standardport which is 1883. 
can you describe me how/where i have to change the mqtt-port in you project to use (for example) 4321?
Thanks a lot!

serb91, 2022-04-14:
Please close it - i edit the MQTT-Port with Gitpod and now i can use your File with a different Port.
Your Manuel also work with my Intex Simple Spa (https://www.steinbach-group.com/de/128482-simple-spa-73976de/)
Thx alot!

closed 2022-04-14 by serb91

jnsbyr, 2022-06-12:
Glad you could help yourself and thanks for the compatibility info. I added it to the readme.

serb91, 2022-06-12:
Thx for your mail
Can you maybe describe me how i can add a static IP?

I search online for an answer but i dont find one 😕

Thx!
Igor

Ps: if its better for you we can write in german 😊

jnsbyr, 2022-06-12:
@serb91 
Your question regarding a static IP is a little of topic - but who cares. Assuming you want you WiFi client to use a static IP you must add a call to WiFi.config() before calling WiFi.begin() in the INO. Have a look at this [tutorial](https://techtutorialsx.com/2020/04/21/esp32-arduino-set-static-ip-address/) for the details. It is about an ESP32 but the Arduino code is the same.

serb91, 2022-10-11:
Sry about off topic..
Thx a lot!
I will look at the tutorial this week