jnsbyr/esp8266-intexsbh20 #6: Node Red Flow

Elektroarzt, 2022-05-07:
Hi Jens,
big congrats to you for your excellent work! I got the PCB with a D1 Mini to work and made an integration in HomeKit via Node Red which is beta at the moment. Did you - by chance - also made some Node Red Flows behind your MQTT Broker? If so, I would be very interested in this for my reference.

As I made a PCB out of your schematic (tested design) I also like to share Fritzing and Gerber data if you like to reshare in your repository for free usage in the community. If you are further interested in a ready-to-go PCB, please let me know and I will send you one or two.

Thanks and best regards,
Thomas

jnsbyr, 2022-05-08:
Hi Thomas,
thanks for taking an interest in my little project from last spring. 
I did not play around with Node Red so your on your own here. I will keep this case open, maybe someone will see your request.
Regarding your PCB design I would prefer your direct participation, if this is OK for you. I already added you as as collaborator to the project. If you go ahead with this idea, you should let me know how much experience you have with Git. If you are not familiar you should
- install the GitHub app on your computer
- checkout the project
- create a new folder named "pcb" on the main level and put the PCB related documents there
- push your update to a new branch, e.g. name it PCB
- create a merge request
If you already know how it works you can commit directly. After your files are included I would like to add some references to your document to the Readme.

Elektroarzt, 2022-05-08:
I'm not a frequent GitHub user but I think I got it, thanks for your quick start guide. Please have a look if everything fits for you and if you need something from me for the readme or so.

Best, Thomas

fizcris, 2022-06-12:
Hey @Elektroarzt could you add links to get the waterproof box? 
Also it would be great to know if you have a PCB lying around so I don't have to manufacture 5pcs to use one.

Elektroarzt, 2022-06-12:
Hi fizcris ... I used the following components for the case:
- Case: Anicoll M686 IP68 (e.g. https://www.amazon.de/gp/product/B075XHW4LS/ref=ppx_yo_dt_b_asin_image_o01_s01?ie=UTF8&th=1)
- Mounting spacer: Standard M3 nylon hex distancer (e.g. https://www.amazon.de/gp/product/B06Y43YY1B/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1), which can be glued in the case easily

As I have a bunch of friends - meanwhile all with the same pool :) - I assembled some pieces as a batch. So unfortunately I don't have a blank PCB but fortunately I have some complete devices left from the batch for self cost (assembled PCB / case / spacer / D1 Mini). If you are interested write a PN under elektroarzt@digital-filestore.de

Regards, Thomas

Elektroarzt, 2022-07-10:
Hi to all pool lovers :)

meanwhile I have a Node Red flow that works like a charm I like to share. It has the following features up to now:
- bridge to Apple HomeKit for remote control via mobile devices (see photo)
- web interface (see photo)
- configurable charts for  pool temperature, ambient temperature, energy cost, heater power
- timer for desired ETA with date and time
- timer for pool chemicals (5in1 chloride tabs)
- timer for automated daily filtering
- failure and status message via telegram

The code is stable but at certain points improvable, e.g. the user interface.
@jnsbyr: if you agree, I would upload the flows

[image]
[image]
[image]

jnsbyr, 2022-07-10:
Looks really good, and I think the timer is a useful extra feature.

Please add a separate subdirectory (e.g. nodered) for the files. If there installation requirements, please add an additional readme into the nodered subdirectory too.

We should also add a new paragraph about the Node-RED support to the Readme. My ideas to put some (or all) of the screenshots above into the assets subdirectory (e.g. with file name prefix NodeRed) or into the nodered subdirectory. Then the Readme can get updated to show the screenshots and give a short introduction.

jnsbyr, 2023-08-11:
@Elektroarzt: If you still think adding the Node Red flow to project is a good idea, please proceed or close this issue.

Elektroarzt, 2023-08-14:
Up to now there were no questions concerning Node Red integration and I feel ioBroker and Home Assistant are the most common ways to implement the pool controller on backend side. Node Red seems a little exotic. So you are absolutely right and I close this issue.

closed 2023-08-14 by Elektroarzt