jnsbyr/esp8266-intexsbh20 #8: Problem command

Drealine, 2022-05-23:
Hi,

I've a problem with the program.
I can view state of each function by manually button but i can't send command. 
MQTT connection is ok.

How can I debug it ?

jnsbyr, 2022-05-24:
It's not easy and depends on the equipment you have available.

1st make sure the ESP is running at 160 MHz. 

If you have a logic analyser or a digi scope, connect it to the ESP output and check if there is any reaction at all and if the timing is as described and shown in the Readme. 

You can also use serial debugging. Use Putty or any other terminal and look at the output after re-enabling the "DEBUG_MSG" commands, especially the ones in SBH20IO.cpp. You will need to study the code a little bit because debugging in time critical section should be avoided or must be very brief. The more serial debugging you enable the less likely it is that sending a command will work at all - but you will be able to see better how it works and later reduce the debug output again.

You may also add a big delay at the point where the output is activated to check with a multimeter if it works at all.

Could also be the bidirectional level shift is not working - or something else. Remote diagnostic is only guesswork from afar.

Elektroarzt, 2022-05-24:
An additional and very simple idea ... did you recognize the difference between published topics (e.g. pool/bubble) and subscribed topics (pool/command/bubble)? If you used publish topics as a command from your software, nothing will happen at the pool.

Drealine, 2022-05-24:
> An additional and very simple idea ... did you recognize the difference between published topics (e.g. pool/bubble) and subscribed topics (pool/command/bubble)? If you used publish topics as a command from your software, nothing will happen at the pool.

Yes, I send command with pool/command/bubble or power etc ..

> It's not easy and depends on the equipment you have available.
>
> 1st make sure the ESP is running at 160 MHz.
>
> If you have a logic analyser or a digi scope, connect it to the ESP output and check if [...]

Ok, I will try with a serial debugging.
But, that's it's strange is that the command work on another SPA. I don't know why ...

I already try to debbugin a little the code by manual put serial.printf message and my MQTT command is correctly detected because the function pressButton is activated in the SBH20IO.cpp.

Drealine, 2022-05-25:
Okay we can solved the issue.
The frequency of the CPU was 80mhz when I flash with the Arduino tool. I've changed to 160mhz and it's ok