jnsbyr/esp8266-intexsbh20 #9: mqtt with authentication (user/pass)

felge1965, 2022-05-24:
Good day,

I need mqtt with authentication (user/pass). Where can I put user/pass in code or config.json?
I use iobroker as mqtt server. I would be grateful for help.

2nd question
Is it possible to change the prefix of the mqtt topic at a central point in the code?
So instead of pool/.. pool0/.. and pool1/..
I have 2 Intex pools in the house.
--- German ----
Guten Tag,

ich benötige mqtt mit authentifizuerung (user/pass). Wo kann ich user/pass im Code oder config.json einfügen?
Ich nutze iobroker als mqtt-Server. Ich wäre dankbar für Hilfe.

2.Frage.
Kann man den prefix des mqtt-Topic an Zentraler Stelle im Code ändern?
Also statt pool/..  pool0/.. und pool1/..
Ich habe 2 Intex-Pools im Haus.

jnsbyr, 2022-05-24:
The pubsubclient MQTT API has an authentication option but so far the project does not use it. Would take a couple of minutes to code and test. Starting point would be _mqttClient.connect_ in MQTTClient.cpp:88. Please consider to extend the code yourself. I would be happy to handle a merge request.

The "pool" topic is defined in common.h:63ff. You could simply compile 2 hardcoded versions. If you want this to be configurable you have the option I mentioned above.

Drealine, 2022-05-25:
@jnsbyr 
I've already code the authentification for MQTT.
Would you make a PR for this ?

jnsbyr, 2022-05-25:
@Drealine
As far as I understand Git, it works like this: you create a new branch for this project, commit it and submit a PR and I will handle the PR and perform the merge with the main branch. If that is OK for you, please go ahead. If you are not allowed to commit or to create a PR let me know. But please be a little patient because I just made sure that I have no compiler near me till next week :-)

Drealine, 2022-05-26:
@felge1965 
Could you please reopen an issue for your second request?
It will be more readable for PR and issues

closed 2022-05-26 by Drealine

jnsbyr, 2022-06-11:
@Drealine 
Thanks for your solution for MQTT authentication. 

I finally found the time to have a look at the code and I have a questions regarding your change to setting the retained flag in most calls to publish. Making the MQTT server retain values might seem to be an advantage. But retaining something old might give a false impression of the current state. So for most values this is not desired from my point of view. As long as there is a connection these values are updated frequently and older values become immediately obsolete.

Why did you change the previous settings?