jnsbyr/esp8266-intexsbh20 #12: Upload

mabeca96, 2022-05-30:
C:\Users\Wemos\Arduino\sketch_may30b\esp8266-intexsbh20\esp8266-intexsbh20.ino: In function 'void setup()':
esp8266-intexsbh20:115:114: error: no matching function for call to 'MQTTClient::setup(const char*, const char [21], const char [11], const char [8])'
  115 |       mqttClient.setup(config.get(CONFIG_TAG::MQTT_SERVER), CONFIG::POOL_MODEL_NAME, MQTT_TOPIC::STATE, "offline");
      |                                                                                                                  ^
In file included from C:\Users\Wemos\Arduino\sketch_may30b\esp8266-intexsbh20\esp8266-intexsbh20.ino:55:
C:\Users\Wemos\Arduino\sketch_may30b\esp8266-intexsbh20\MQTTClient.h:50:8: note: candidate: 'void MQTTClient::setup(const char*, const char*, const char*, const char*, const char*, const char*)'
   50 |   void setup(const char* mqttServer, const char* mqttUsername, const char* mqttPassword,const char* clientId, const char* willTopic, const char* willMessage);
      |        ^~~~~
C:\Users\Wemos\Arduino\sketch_may30b\esp8266-intexsbh20\MQTTClient.h:50:8: note:   candidate expects 6 arguments, 4 provided
Using library ArduinoJson at version 6.19.4 in folder: C:\Users\Wemos\Arduino\libraries\ArduinoJson 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Wemos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi 
Using library PubSubClient at version 2.8 in folder: C:\Users\Wemos\Arduino\libraries\PubSubClient 
Using library LittleFS at version 0.1.0 in folder: C:\Users\Wemos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\LittleFS 
Using library ESP8266httpUpdate at version 1.3 in folder: C:\Users\Wemos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266httpUpdate 
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\Wemos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266HTTPClient 
exit status 1
no matching function for call to 'MQTTClient::setup(const char*, const char [21], const char [11], const char [8])'

jnsbyr, 2022-06-11:
@mabeca96 
should be fixed with the patch from @piitaya, try again

closed 2022-06-11 by jnsbyr

mabeca96, 2022-06-15:
Hello, yes the problem is fixed, but a new one is coming. When starting, the message cofig.json would not be found. I placed the file in the Data directory. Where do I have to store the file

jnsbyr, 2022-06-15:
Placing the config file into the data subdirectory is not enough. You must also upload this file to the filesystem on the ESP with a tool of your choice - the Arduino IDE will not do this automatically. Please refer to the description under "Configuration". I recommend using the "Arduino ESP8266 LittleFS Filesystem Uploader". It will extend the Arduino IDE tools menu with 2 new commands.

mabeca96, 2022-06-19:
OK
I was able to upload the config.
now I get the following message
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000689b0
~ld
SB-H20 WiFi Controller 1.0.2.0
build with Arduino Core for ESP8266 3.0.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)
error parsing config file: NoMemory

jnsbyr, 2022-06-19:
Please check if your config file:
- does not have any JSON syntax errors
- has less than 256 bytes
If it has nearly or more than 256 bytes, remove some of the optional entries for testing purposes and try again.

mabeca96, 2022-06-19:
Thank you very much for the support.
The connection to the MQQT is stable.

jnsbyr, 2022-06-19:
Happy to hear! Please be so kind and elaborate a little bit. What did you change so that it start working?

PR #14 (2022-06-11): add missing mqtt user and password parameter
Hello 🙂 
The username and password parameters were missing.
Fixes #12