jnsbyr/esp8266-intexsbh20 #13: question

Monsardi, 2022-06-06:
Hello,
I have a intex sjb hs with salt electrolyse + jet streams.
Do you think it is possible to add this functionality?

all the best Peter

jnsbyr, 2022-06-11:
@Monsardi 
The display of the SJB-HS looks different than that of the SBH-20, e.g. it has additional buttons. Without getting my hands on the hardware and spending some time with it it is impossible to say for sure.

Please have a look at the project [DIYSCIP](https://github.com/yorffoeg/diyscip) because it claims to support SSP-xxx and SJB-xxx models.

mabeca96, 2022-07-10:
Hello
I also have an Intex SJB.
I connected the esp8266-intexsbh20 project to the pool.
I just have the problem that I can't turn the filter on and off.
When I turn on the filter via ESP, the bubble turns on.
The info what on/off is correct.
Also the temperature is displayed correctly

jnsbyr, 2022-07-10:
@mabeca96: That is very interesting. I did not expect that it would work at all. 

On the assumption that the control principle of my project also works with the SJB model, some changes are required to address the different mapping of the control function like the bubbles. The approach would be:

- add a new config option to select the Intex model (SBH, SJB)
- support different control bit mappings for each supported model 

Because I do not have a SJB I would prefer is someone changes the code, makes tests and submits a merge request. Otherwise I would have to dry code and you could test. My current lack of spare time will not allow me to do the latter, at least not before the summer is over.

Luckily there is another option: you could hard code the changes yourself. That way we will quickly know if it really works. Open SBH20IO.cpp and change the values in the enum _FRAME_BUTTON_. Look into the project [DIYSCIP](https://github.com/YorffoeG/diyscip/blob/master/src/CTRLPanel.cpp) to get the correct values or just try out different values. You already know that the current value of the filter is your value for the bubbles.

Let me know if you can make it work this way and if yes, post the details of the changes you made.

mabeca96, 2022-07-11:
I will try it with hard code.
Can you help me see the value of DIYSCIP?
Example 
You: const uint16 POWER = 0x0400
DIYSCIP: BUTTON_POWER U5 | COMMAND_ADDRESS_S2
BUTTON_POWER COMMAND_ADDRESS_S0
#define U5 0x80
#define COMMAND_ADDRESS_S4 0x04

Monsardi, 2022-07-11:
Thanks for jumping in the conversation. I can confirm that I have the same buttons/functions that work and the same that fail.
Pushing the bubble function triggers the jetstreams and the filter is triggered by the bubble button.
Of course there's also nothing that triggers the salt electrolyse.
I'm curious whether the hard code road will work!

jnsbyr, 2022-07-13:
When I look into the DIYSCP code I don't see a definition for a bubble or a jet button and there seems to be no additional documentation other than the code and the schematic. So I don't have anything to compare or recommend.

Please assign an otherwise unused 4digit hex value that sets only 1 bit (1, 2, 4, 8) to one or more of the functions of FRAME_BUTTON (e.g. 0x0100) which do not work for you, build and try these functions. Should be around 16 possibilities - so this should not take too long. Either it helps or it doesn't. Beware that this way you might be able to send a code that triggers a function reserved by the manufacturer that cannot be triggered by the real panel.

Regarding functions the SBH model does not support like the electrolyse - if you can find the button code and it works consistently for you than let me know.

mabeca96, 2022-07-25:
I found out the following code

  const uint16 POWER     = 0x0400;
  const uint16 FILTER    = 0x0080;
  const uint16 HEATER    = 0x8000;
  const uint16 DESIFECT  = 0x0001;
  const uint16 TEMP_UP   = 0x1000;
  const uint16 TEMP_DOWN = 0x2000;
  const uint16 TEMP_UNIT = 0x2000;
//  const uint16 JetDuesen       = 0x0008;
//  const uint16 JetBubble       = 0x0002;
//  const uint16 F-C     = 0x2000;
//  const uint16 JetBubble       = 0x0002;

what I am still missing is the display whether the disinfection is on or not.
Can someone explain to me where I have to set additional entries to control them from the MQTT (e.g. disinfection)

Monsardi, 2022-08-05:
have you found out the missing entries?
Unfortunately I have no knowledge at all aboit coding so I can't be of any help.

jnsbyr, 2022-09-05:
> I found out the following code
Did you try the setting you posted by changing FRAME_BUTTON in SBH20IO.cpp? 

If yes, please elaborate your results a little bit. Your current listing has some contradicting lines e.g. TEMP_DOWN and TEMP_UNIT are both 0x2000. Can it be that you do not have a TEMP_UNIT function? On the other hand you added some extra values. I assume these are the extra function of the SJB. What is "F-C"?

 As soon as I have the hole picture I can add a variant for the SJB model and the necessary command support via MQTT.

Monsardi, 2022-09-08:
F-C switches the display from fahrenheit to Celsius

jnsbyr, 2022-09-17:
@Monsardi 
Thanks for your reply. So "F-C" is the same as "TEMP_UNIT" of the SBH - no big surprise here. That leaves the contradiction that your "TEMP_DOWN" has the same value.

Could you please answer all questions from my last comment in some detail? I do not have a SJB - so without you input I cannot make the required changes.

Monsardi, 2022-09-17:
Hi, i'm afraid I can't answer your other questions, as it is maceba96 who found out the rest of the code.

jnsbyr, 2022-09-17:
@Monsardi 
Thanks for the quick feedback. Sorry I mixed up who had posted the codes originally. I still hope that @mabeca96 will add a comment.

jnsbyr, 2023-03-03:
@FireSale14

@mabeca96 has reported several codes that work with the SJB (see https://github.com/jnsbyr/esp8266-intexsbh20/issues/13#issuecomment-1194086643). So the hardware and the software of this project seem to be compatible. Now someone with an SJB needs to continue the code analysis, either with a logic analyser or by testing individual codes (see https://github.com/jnsbyr/esp8266-intexsbh20/issues/13#issuecomment-1183525195). 3rd alternative is to try reverse engineering the [DIYSCIP](https://github.com/YorffoeG/diyscip/) code.

Once this analysis is complete the support for the SJB can be added with a new configuration option. The coding could be done by me or can be provided as a merge request.

Please note that I am still unsure if finding the right codes is all it needs. With a working SJB and a logic analyser finding out is not that hard except for the wiring if you don't want to cut the display cable.

FireSale14, 2023-05-09:
@jnsbyr 

Can you please help me? What hardware do I need? This? https://amzn.eu/d/cYtB7Zk? Which wires of the display should I connect it to? What should I press so that you can do something with it?
Thank you very much.

jnsbyr, 2023-05-09:
I will be happy to include anything you can find out. Maybe you need no logic analyser for this. 

First look at [comment #13](https://github.com/jnsbyr/esp8266-intexsbh20/issues/13#issuecomment-1194086643). It already contains more than half of the solution. Use it as a starting point and confirm what @mabeca96 has documented. 

For the implementation of a new model it should first be verified that the communication principle of the SB-H20 works at all, by getting any positive reaction to commands and a feedback of a status change using the current firmware.

If this is the case, the definitions of *FRAME_XXXXX* constants in [SBH20IO.cpp](https://github.com/jnsbyr/esp8266-intexsbh20/tree/master/SBH20IO.cpp) need to be adjusted, maybe also *DIGIT* and *ERROR*. If possible *FRAME_BUTTON* and *FRAME_LED* should be completed first.

As the SJB-HS has more functions than the SB-H20 you will probably need to select one of the button function to test for the missing commands by changing the values and do something similar with the LED inputs. Typically only 1 bit needs to be set in a command, so there are not that may test that need to be done.

In a second step one would add the additional code needed to support the extra functions and add a config entry to make the model  selectable. For this seconds step no hardware is needed, so I could cover this or assist.

You may also have look at the documentation at the top of [SBH20IO.h](https://github.com/jnsbyr/esp8266-intexsbh20/tree/master/SBH20IO.h), to get some background, what this is good for.

Please consider that it might not work. Then you may have learnt something but you have also permanently modified the pool and you need to find other uses for the ESP8266 module. Be sure you have a little experience with DIY electronics. Otherwise the pool controller might need replacement, too.

If you don't want to build the ES8266 module first or if the tests described above do net get any useful results the logic analyser comes into play. Here is another [option](https://www.azdelivery.de/en/products/saleae-logic-analyzer). Look at the logic analyser plot and the schematic - the name of the pins to connect to are on the left side of the plot. Having a little experience with this topic is recommended - otherwise you will have a steep learning curve ahead of you. The task it to make recordings of the various operations and extract the interesting sections from the tons of data you will get. Will probably take several days to set it up and get it done.

algermi, 2023-05-24:
Hello all,

I have today also thanks Thomas my ESP8266 connected to my SJB-HS-20. Can someone tell me if I operate the control via HomeKit, then it works for me only with the Subscribed Topics. 
If I understand correctly, I stand with the Subscribed Topics the Whirlpool and with the Published Topics I query the current values? Right?
I am a newbie with MQTT.

Has now already someone managed to find out the values for SJB-HS-20 for only "filtering" and "salt water treatment". 
Maybe someone from Germany is out here and has the same challenges.
I would be very happy about a reply.

Translated with www.DeepL.com/Translator (free version)

jnsbyr, 2023-05-24:
@algermi 

Can't help you much with HomeKit, but you basic understanding of MQTT seems to be right. If a node "publishes" something, that's the output to the MQTT server. And "subscribing" is the other direction where a node defines the topics it is interested in.

From you other comment I take that the project basically works with the SJB-HS-20. Please don't wait for someone to improve it but try it yourself, because there are not that many users who have your setup. Use my previous comment as a starting point, select any entry of *FRAME_BUTTON*, e.g. "HEATER", modify the current code value to a previously unused value and test it by modifying  "HEATER" via MQTT. Use codes where only 1 bit is set - there are not that many. This is the easiest way to find the missing mappings.

The SBH20 has 7 buttons, the SJB-HS has 9, with the extra functions JET and DISINFECTION. @mabeca96 has mentioned above that FILTER is 0x0080 and DISINFECTION is 0x0001, so it looks like only JET is missing. But some other mappings are not unique and need to be clarified. This is the current state of the mapping:

**CODE FUNCTION**
0x0001 DISINFECTION
0x0002 FILTER ?
0x0004
0x0008 BUBBLE ?
0x0010
0x0020
0x0040
0x0080 FILTER / TEMP_DOWN ?
0x0100
0x0200
0x0400 POWER
0x0800
0x1000 TEMP_UP / TEMP_UNIT ?
0x2000 TEMP_UNIT ?
0x4000
0x8000 HEATER
0x????  JET

Please give it a try and publish your test results.

algermi, 2023-05-25:
Hi Jens,

ca we speek in German. If you want we can also communicate by mail or whatsapp's. of course we share the obtained results here in english

FireSale14, 2023-05-26:
Super, 
sobald es hier mal bissel besseres Wetter ist werde ich den Pool aufbauen und dann durch testen.

RianFlow, 2023-05-26:
Hi. Sry wenn ich so dazwischen grätsche, aber hat einer von euch vielleicht eine genau Anleitung welche Dateien ich wo genau hin installieren muss auf den ESP? Und wie es mit ESPHome genau abläuft? Ich würde gerne zum steuern Home Assistant nutzen. 
Meine Platine habe ich sonst soweit fertig, wäre der letzte schritt der mir noch fehlt!
Schon mal danke im Vorraus

algermi, 2023-05-26:
Servus zusammen,

also ich hab jetzt mal etwas rum probiert und das Ergebnis sieht folgendermaßen aus.

folgende Werte sind für die Steuerung relevant:

0x0001 DISINFECTION ##getestet
 0x0002 BUBBLE  ##getestet
 0x0004
 0x0008 JET  ##getestet
 0x0010
 0x0020
 0x0040
 0x0080 FILTER  ##getestet
0x0100
 0x0200
 0x0400 POWER  ##getestet
 0x0800
 0x1000 TEMP_UP / TEMP_UNIT ? 
0x2000 TEMP_UNIT ?
 0x4000
 0x8000 HEATER  ##getestet

Die Funktionen wurden alle von mir erfolgreich getestet. 
Was aktuelle eben noch nicht geht ist die Einstellung der Temperatur.
Die Abfrage der aktuellen Temperatur funktioniert auch. 
was ich nicht hinbekomme ist die Steuerung über HomeKit als Thermostat um damit die Temperatur rauf oder runter zu stellen.

Kann mir jemand sagen, wie ich manuell also z.B. per Commandzeile MQTT Befehle absetzen kann. Vielleicht stimmt ja meine HomeKit Konfiguration für das Thermostat noch nicht.

english:

Hello all,

I have now tried something and the result looks like this.

The following values are relevant for the control:

0x0001 DISINFECTION ##tested
 0x0002 BUBBLE ##tested
 0x0004
 0x0008 JET ##tested
 0x0010
 0x0020
 0x0040
 0x0080 FILTER ##tested
0x0100
 0x0200
 0x0400 POWER ##tested
 0x0800
 0x1000 TEMP_UP / TEMP_UNIT ? 
0x2000 TEMP_UNIT ?
 0x4000
 0x8000 HEATER ##tested

The functions have all been tested successfully by me. 
What is currently not working is the setting of the temperature.
The query of the current temperature also works. 
What I can't manage is the control via HomeKit as a thermostat to set the temperature up or down.

Can anyone tell me how I can send MQTT commands manually, e.g. via command line. Perhaps my HomeKit configuration for the thermostat is not yet correct.

algermi, 2023-05-26:
@jnsbyr 
Servus Jens, ich hoffe Du kannst mit den Info's etwas anfangen.
Meinst Du könntest bitte die neuen Funktionen mit aufnehmen. 

Hi Jens, I hope you can do something with the information.
Do you think you could please include the new functions?

algermi, 2023-05-26:
Im ioBroker stehe ich alle Werte, auch die eingestellte und aktuelle Pooltemperatur

[image]

Elektroarzt, 2023-05-26:
> Can anyone tell me how I can send MQTT commands manually, e.g. via command line. Perhaps my HomeKit configuration for the thermostat is not yet correct.

You can use the app MQTT Analyzer for iOS Devices for example or MQTT Explorer for MacOS. Both really suitable tools. Just add your Broker and subscribe to your pool topics and you're ready to rock'n roll

jnsbyr, 2023-05-26:
@algermi 

Danke, dass du die Zuordnung der Button-Codes für SJB-HS ausgetestet hast.

0x0001 DISINFECTION ##tested
 0x0002 BUBBLE ##tested
 0x0004
 0x0008 JET ##tested
 0x0010
 0x0020
 0x0040
 0x0080 FILTER ##tested
0x0100
 0x0200
 0x0400 POWER ##tested
 0x0800
 0x1000 TEMP_UP / TEMP_UNIT ? 
0x2000 TEMP_UNIT ?
 0x4000
 0x8000 HEATER ##tested

Ich werde eine neue Version erstellen, die eine experimentelle Unterstützung für SJB-HS hat. Vielleicht gelingt das bis nächste Woche.

Wäre noch zu klären, warum die Temperatureinstellung nicht funktioniert. Aus den älteren Kommentar ist nicht klar zu erkennen, ob es schon mal jemand geschafft hat. Selbst wenn sich die Temperatur nicht ändert, sollte die Steuerung kurz piepen, wenn du den Code für die Temperaturänderung schickst, so als ob du es direkt auf dem Display bedienst. Möglicherweise fängt dann auch der Wert im Display an zu blinken. Wenn das passiert, hast du schon mal die richtigen Code gefunden. Das Verstellen selbst ist ein zeitkritischer Ablauf. Vielleicht braucht der SJB-HS dafür ein angepasstes Timing, aber das wäre der übernächste Schritt.

Hast du versucht den Code für die Umschaltung °C/F zu finden? Denn der fehlt leider auch noch.

algermi, 2023-05-27:
Moin, was ich gerade noch auf die schnelle rausgefunden habe,

Temperatur UP funktioniert, so wenn z.B. 35 Grad auf 38 Grad erhöht werden soll.
Temperatur Down, z.B. von 38 auf 35 Grad stellt das Display hin und her von Grad auf Fahrenheit.

Also hätten wir das auch schon, jetzt fehlt nur noch der richtige Wert für Temperatur Down.

en:

What I just found out at the last minute,

Temperature UP works, e.g. when 35 degrees is to be increased to 38 degrees.
Temperature Down, e.g. from 38 to 35 degrees, switches the display back and forth from degrees to Fahrenheit.

So we've got that, now we just need the right value for Temperature Down.

jnsbyr, 2023-05-27:
@algermi 

Super, das sind echte Fortschritte. 

Habe mit der Programmierung begonnen. Das Projekt komplett auf SJB-HS umzubauen wäre relativ einfach. Aber es soll beide Intex Modelle per Konfiguration unterstützen, und das ist dann schon etwas mehr Aufwand - vor allem da der Code auch relativ performant bleiben muss.

Habe auch festgestellt, dass noch das Mapping für *FRAME_LED* ermittelt werden muss, denn schließlich will man über MQTT ja auch die Rückmeldung, ob Heizung, Jet, etc. an oder aus sind. Dazu bitte am Display die Funktionen ein- und ausschalten und über MQTT beobachten, welcher Wert sich ändert. Die möglichen Codes sind die Gleichen wie bei *FRAME_BUTTON*.

jnsbyr, 2023-05-27:
@algermi

Eine Analyse für *FRAME_LED* ist nicht mehr erforderlich. Habe die Zuordnung in [DIYSCIP](https://github.com/YorffoeG/diyscip/blob/master/src/CTRLPanel.cpp) gefunden - die Werte sind direkt kompatibel. Auch den fehlenden Wert für *TEMP_DOWN* in  *FRAME_BUTTON* konnte ich umrechnen. 

Nur *TEMP_UNIT* (Umschaltung °C/F) fehlt noch, da es bei DIYSCIP nicht verwendet wird. Sobald du den Code meldest, der bei dir funktioniert, kann ich eine neue Firmware fertig machen, die SJB-HS unterstützt.

Was im Projekt noch ganz fehlt ist das Lesen und Schreiben der Desinfektionszeit bei SJB-HS. Aktuell kann ich noch nicht sagen, ob und wann das unterstützt wird.

algermi, 2023-05-28:
@jnsbyr 
Guten Morgen Jens, der Wert für die Umschaltung F/C sollte ja = 0x2000 sein.

Für was steht in der Standard Konfiguration der Wert TEMP_UNIT ? Ist das bei SBH20 auch die Umschaltung von F/C?
Oder für was steht TEMP_UNIT;

Wie bereits geschrieben, wenn ich mit deinem Setup bei mir mit deinem Setup und dem Command pool/command/water/tempSet die Temperatur von z.B. 38 auf 30 stellen will, dann piept der Whirlpool
8 mal. Aber er schaltet halt 8 mal zwischen C und F um.

Hoffe das hilft Dir weiter.

Magst Du mir mal den Wert für TEMP_DOWN zukommen lassen, den DU herausgefunden hast. DANKE

algermi, 2023-05-28:
Jetzt hätte ich nochmal eine Frage an die HomeKit Leute. Der Heizzustand ist ja on/off/standby.

So kann ich es auch im IOBROKER sehen. Jemand eine Idee wie man das im HomeKit visualisieren kann. 

Standardmäßig wenn ich die Funktion als Switch einbaue, dann hab ich ja nur  die Möglichkeit ON/OFF

en:

Now I have another question for the HomeKit people. The heating status is on/off/standby.

I can also see it in the IOBROKER. Does anyone have an idea how to visualise this in HomeKit? 

By default, when I install the function as a switch, I only have the option ON/OFF.

jnsbyr, 2023-05-28:
@algermi 

> Magst Du mir mal den Wert für TEMP_DOWN zukommen lassen, den DU herausgefunden hast. 

Sehr gern, wäre toll wenn du da noch mal einen Blick 'drauf werfen könntest, bevor ich eine Firmware fertig mache:
```
namespace FRAME_BUTTON
{
  const uint16 DISINFECTION = 0x0001;
  const uint16 BUBBLE       = 0x0002;
  const uint16 JET          = 0x0008;
  const uint16 FILTER       = 0x0080;
  const uint16 POWER        = 0x0400;
  const uint16 TEMP_UP      = 0x1000;
  const uint16 TEMP_DOWN    = 0x0200;
  const uint16 TEMP_UNIT    = 0x2000;
  const uint16 HEATER       = 0x8000;
}

namespace FRAME_LED
{
  const uint16 POWER          = 0x0001;
  const uint16 BUBBLE         = 0x0002;  // max. 30 min
  const uint16 HEATER_ON      = 0x0080;  // max. 72 h, will start filter, will not stop filter
  const uint16 NO_BEEP        = 0x0100;
  const uint16 HEATER_STANDBY = 0x0200;
  const uint16 JET            = 0x0400;
  const uint16 FILTER         = 0x1000;  // max. 24 h
  const uint16 DISINFECTION   = 0x2000;  // max. 8 h
}
```

algermi, 2023-05-28:
@jnsbyr 

servus, klar ich teste das mal durch. ich werde aber erst morgen dazukommen.

algermi, 2023-05-28:
@jnsbyr 

Ich konnte es doch nicht lassen, also TEMP_DOWN 0x0200 ist korrekt und funktioniert nun auch.

en:

I couldn't leave it alone after all, so TEMP_DOWN 0x0200 is correct and now also works.

jnsbyr, 2023-05-28:
@algermi 
Vielen Dank für die Unterstützung. Dafür gibt es nun eine Firmware-Version zu testen, wo alle Funktionen gleichzeitig untersützt werden.

Experimental support for the Intex PureSpa SJB-HS has been added to the project and the new version 1.0.3.0 is available in the _development_ branch for testing, see commit c185f1f, download [here](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop).

Changes:
- no default PureSpa model, comment in the desired model variant at the top of [common.h](https://github.com/jnsbyr/esp8266-intexsbh20/blob/develop/common.h) before building the firmware
- class SBH20IO renamed to PureSpaIO

Notes:
- reading and writing the disinfection time is not implemented (yet)
- the hardware and firmware will probably support the Intex PureSpa SSP-H models with minor modifications

algermi, 2023-05-30:
Moin Jens, 

ich hab jetzt mal die neue Firmware drauf. Vorher die Datei common.h

wie folgt angepasst:

Auszug aus der common.h

#ifndef COMMON_H
#define COMMON_H

#include <climits>
#include <../d1_mini/pins_arduino.h>

// select Intex PureSpa model by commenting in the desired variant
//#define MODEL_SB_H20
#define MODEL_SJB_HS

//#define SERIAL_DEBUG

Fehlermeldung im Serialmonitoring:

rf[112] : 0⸮unsupported Intex PureSpa model MQTT WiFi 
Controller 1.0.3.0
build with Arduino Core for ESP8266 3.1.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)
config file loaded successfully (has 8 entries)
trying to connect to MQTT server ... 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021f312 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40251bf4 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd60 end: 3fffffd0 offset: 0150
3ffffeb0:  514d0400 3f045454 3ffeef00 00000030  
3ffffec0:  00000000 40251bf4 00000000 00000001  
3ffffed0:  00000001 00000100 00000020 4010146c  
3ffffee0:  3fff0bf4 00000010 3fff0978 3ffeef00  
3ffffef0:  00000000 3ffef008 3ffeeb50 4020a8f0  
3fffff00:  00000001 3ffe8952 00000001 3ffeef00  
3fffff10:  00000000 3ffef008 3ffeeb50 40206731  
3fffff20:  00000001 3ffe8952 3fff0978 40210a15  
3fffff30:  00001148 00000229 3ffe8630 401010ea  
3fffff40:  00000000 00000f86 87ef9db2 0011f38d  
3fffff50:  00000000 3ffeeb50 3ffeeb50 3ffeef00  
3fffff60:  3ffeeb18 3ffeeb50 3ffeeb50 4020693a  
3fffff70:  3fffdad0 00000000 3ffeec38 40208791  
3fffff80:  00000000 000c000f 00000000 40107924  
3fffff90:  2404a8c0 feefeffe feefeffe 3ffef194  
3fffffa0:  3fffdad0 00000000 3ffef168 3ffef194  
3fffffb0:  3fffdad0 00000000 3ffef168 40211e48  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101711  
<<<stack<<<

rf[112] : 0⸮unsupported Intex PureSpa model MQTT WiFi Controller 1.0.3.0
build with Arduino Core for ESP8266 3.1.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)
config file loaded successfully (has 8 entries)
trying to connect to MQTT server ... 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021f312 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40251bf4 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd60 end: 3fffffd0 offset: 0150
3ffffeb0:  514d0400 3f045454 3ffeef00 00000030  
3ffffec0:  00000000 40251bf4 00000000 00000001  
3ffffed0:  00000001 00000100 00000020 4010146c  
3ffffee0:  3fff0bf4 00000010 3fff0978 3ffeef00  
3ffffef0:  00000000 3ffef008 3ffeeb50 4020a8f0  
3fffff00:  00000001 3ffe8952 00000001 3ffeef00  
3fffff10:  00000000 3ffef008 3ffeeb50 40206731  
3fffff20:  00000001 3ffe8952 3fff0978 40210a15  
3fffff30:  00001148 00000229 3ffe8630 401010ea  
3fffff40:  00000000 00000f86 87ef9db2 0011f38d  
3fffff50:  00000000 3ffeeb50 3ffeeb50 3ffeef00  
3fffff60:  3ffeeb18 3ffeeb50 3ffeeb50 4020693a  
3fffff70:  3fffdad0 00000000 3ffeec38 40208791  
3fffff80:  00000000 000c000f 00000000 40107924  
3fffff90:  2404a8c0 feefeffe feefeffe 3ffef194  
3fffffa0:  3fffdad0 00000000 3ffef168 3ffef194  
3fffffb0:  3fffdad0 00000000 3ffef168 40211e48  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101711  
<<<stack<<<

algermi, 2023-05-30:
Servus Jens,

also erster Funktionstest abgeschlossen. Danke für's nachbessern des Scripts.

Model wird richtig erkannt. Intex PureSpa SJB-HS

pool/command/jet geht nur on, off geht nicht. 2. mal on senden, dann geht der Jet wieder aus, bei Bubble geht on/off.

pool/command/disinfection on, dann schaltet er auf 3 Stunden Desinfektion. Mehrmals "on" senden bringt keine Veränderung. Also aktuell keine Auswahl 3/5/8 Stunden.

Heater On/Off geht.

Filter On/Off geht

Mega Job, Geiler Schei... .

Mit Dir kann man arbeiten. :-)

algermi, 2023-05-30:
@jnsbyr 

was ich gerade noch bemerkt habe, bei Filter=on, da schickt er anscheinend gleich wieder ein "off" hinterher.
Filter bleibt zwar an, aber im MQTT Server sehe ich folgenden Eintrag.
Weiss jetzt nicht ob das von Dir kommt, ich sehe auch im HomeKit, wenn ich den Schalter anschalte nur kurz on, dann springt der auf off.

May 30 15:59:25 raspberrypi homebridge[596]: [30.5.2023, 15:59:25] [Filter] Received MQTT: pool/filter = on
May 30 15:59:25 raspberrypi homebridge[596]: [30.5.2023, 15:59:25] [Filter] Received MQTT: pool/filter = off

jnsbyr, 2023-05-30:
@algermi 

> Mit Dir kann man arbeiten.

Dito. Da ich kein SJB-HS habe und mir wohl auch keiner seins für die Entwicklung vorbeibringt, geht so zusammen trotzdem :-)

> ... bei Filter=on, da schickt er anscheinend gleich wieder ein "off" hinterher.

Habe noch einen Copy&Past-Bug gefunden, wo der Zustand der Desinfektion als Filterzustand gemeldet wurde. Neue Version 1.0.3.2 verfügbar. Wahrscheinlich ist das der Grund.

> Auch wenn ich den Heater anschalte, dann geht ja automatisch der Filter mit an ...

Das macht die Intex-Steuerung.

> ... das kann ich auch nicht erkennen im MQTT Client, da bleibt Filter auf Off.

Sollte eigentlich nicht so sein. Nimm die neue Version, schalte manuell am Display den Filter ein- und aus und beobachte, ob das per MQTT gemeldet wird.

> pool/command/jet geht nur on, off geht nicht. 2. mal on senden, dann geht der Jet wieder aus

Das kann ich leider nicht seblst nachvollziehen. Das erfolgreiche Umschalten hängt u.a. von der LED-Zuordnung ab. Schalte noch mal manuell am Display den Jet ein- und aus und beobachte, ob jede Änderung per MQTT gemeldet wird.

> pool/command/disinfection on, dann schaltet er auf 3 Stunden Desinfektion. Mehrmals "on" senden bringt keine Veränderung. Also aktuell keine Auswahl 3/5/8 Stunden.

Habe inzwischen einen möglichen Lösungsansatz, aber der ist relativ aufwendig. Würde gern erst die anderen Punkte oben alle abgefrühstückt haben. Du solltest die Zeit manuell am Display einstellen können. Bitte Info, ob sich die Intex-Steuerung beim Neustart der Desinfektion den zuletzt verwendeten Wert merkt oder z.B. immer 3 h vorschlägt.

algermi, 2023-05-30:
@jnsbyr 

Servus, 
ich werd morgen mal die neue Firmware testen, wie bereits erwähnt steht Dir mein SJB-HS gerne Remote zur Verfügung. Denke aber wir bekommen das final noch gelöst. Schönen Abend noch.

Lg Alex

algermi, 2023-05-31:
@jnsbyr 

Moin Jens, also das mit dem Filter-Thema ist nun gefixt. 

JET-Thema ist noch offen. Wenn ich per Display JET ON/OFF schalte, wir im MQTT sauber mit protokolliert. (pool/jet = on oder off)

wenn ich

pool/command/jet On/Off senden, da greift das Kommando (command) aber scheinbar wird der Topic Pool/JET nicht aktualisiert.

Zu diesem Thema:
(Bitte Info, ob sich die Intex-Steuerung beim Neustart der Desinfektion den zuletzt verwendeten Wert merkt oder z.B. immer 3 h vorschlägt.)

Beim drücken des Buttons Desinfektion wird immer als erstes der Wert 3 h vorgeschlagen, erneutes drücken des Buttons ändert dann auf 5 h und nochmaliges drücken auf 8 h.

algermi, 2023-06-01:
@jnsbyr 

Moin, genau richtig, die Hoch/Runter-Buttons spielen für die Desinfektion keine Rolle.

Wenn der Desinfektionsbutton 4 mal hintereinander gedrückt wird ist die Desinfektion "wieder deaktiviert" d.h. das Desinfektionsprogramm startet nicht. Wenn das Desinfektionsprogramm z.B. 3 Stunden läuft und man es abschalten will, so muss man 4 mal drücken, läuft das Desinfektionsprogramm für 5 Stunden und soll beendet werden, dann muss man 3 mal drücken usw. 

Die neue Version werde ich mir gleich mal testen und gebe dir wieder bescheid.

algermi, 2023-06-01:
@jnsbyr 

Moin nochmal. Das Thema Jet ist Erld. on/off command funktioniert.

jnsbyr, 2023-06-01:
Super, danke fürs testen. 👍 Werde die aktuelle Version als "stabil" freigeben,

Mit deinen Rückmeldungen zur Bedienung der Desinfektion werde ich mir eine Lösung überlegen. Diese Erweiterung erfordert allerdings größere Umbaumaßnahmen an einer anderen Stelle, deshalb werde ich mir dafür ein paar Wochen Zeit nehmen.

algermi, 2023-06-02:
@jnsbyr 
Moin Jens, denke das mit der Desinfektion ist nicht zeitkritisch, wichtig ist, das sie überhaupt läuft. Danke Dir für Deine Bemühungen

mabeca96, 2023-06-04:
Hallo,

Gibt es eine Anleitung, wie ich die Firmware mit Arduino Flaschen kann?
Ich habe mir leider die Konfiguration zerschossen.

jnsbyr, 2023-06-04:
@mabeca96 

Bitte folge den im [README unter Firmware](https://github.com/jnsbyr/esp8266-intexsbh20#firmware) beschriebenen Hinweisen. 

Zusätzlich brauchst du einen USB/Seriell-Adapter. Wie man den an den ESP8266 anschließt, ist schon sehr oft an anderer Stelle in verschiedenen Sprachen beschrieben worden, nimm einfach deine Lieblings-Suchmaschine. Wenn die Verkabelung steht und die Firmware mit der Arduino IDE erfolgreich kompiliert wurde, gib es da auch noch einen "Hochladen" Button. 

Wenn das nicht reichen sollte, dann versuche etwas genauer zu beschreiben, womit du Schwierigkeiten hast.

FireSale14, 2023-06-11:
Hi,
erstmal viele lieben dank für eure Mühen. Wirklich tolle Arbeit.
Ich habe meinen SJB nun auch endlich in Betrieb (hat ein loch). 
Funktioniert auch eigentlich alles. Der ESP verbindet sich mit dem WLAN und dem MQTT Server. Leider verbindet er sich ca. alle 5 Sekunden neu mit dem MQTT Server. Ich habe auch schon einen neuen ESP probiert und einen Hardware defekt auszuschließen. So lang der ESP nicht mit dem Pool verbunden ist funktioniert auch "alles". Er verbindet sich ganz normal. Verbindet man die Datenleitungen beginnt es das er sich alle paar Sekunden mit dem Server neu verbindet.
Hat jemand eine Idee?

Vielen dank

jnsbyr, 2023-06-11:
@FireSale14 
Even if its not easy, try English, you will probably reach more people who could help.

First make sure you use the version 1.0.3.3 form the main branch.

From your description it might be possible that the MQTT server immediately tries to change the temperature when connecting and that fails somehow. You should know that the firmware always needs to disconnects for this, but typically this happens only once.

You might get more infos if you enable the debug output by commenting in the define at the top of common.h. Have a look for yourself and if it does not tell you anything, post the most interesting part, preferably as a new issue.

jnsbyr, 2023-06-18:
@algermi & all other SJB-HS users

There is a new firmware version 1.0.5.1 in branch [develop](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop), commit 85007220, that supports reading the disinfection time (see updated [README](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop#mqtt)).

Before I will make this an official release I would prefer to get your feedback if it is working as expected, because I do not have a SJB-HS to test it myself.

SpaWnOL68, 2023-06-21:
@jnsbyr 
I tried the firmware version 1.0.5.1 and i can have now the disinfection time. The switch seems to work in HA but the state is UNKNOWN for pool/command/disinfection

jnsbyr, 2023-06-21:
@SpaWnOL68 
Thanks for your feedback. 

Reading the disinfection time and writing the disinfection command are separate things. The new firmware "only" improves on the reading to find out, if it works at all. 

With your test result I will be able to make a modification for the disinfection command. It currently supports the values on/off but will actually toggle through the available time settings, right? The next version will probably be able to set a specific disinfection time.

SpaWnOL68, 2023-06-21:
> It currently supports the values on/off but will actually toggle through the available time settings, right? 

Yes, exactly. 0 = Off and 3/5/8 = On 
Great improvement, thanks !
Tell me when the new version is available

jnsbyr, 2023-06-23:
There is a new firmware version 1.0.5.2 in branch [develop](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop), commit ebe17e53d73a85bc9919bbc3bfe0a023fd7addce, that supports setting the disinfection time (see updated [README](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop#mqtt)).

The MQTT disinfection command now requires hours (0,3,5,8) instead of on/off.

Please test and report what happens when you use this new feature.

SpaWnOL68, 2023-07-08:
> There is a new firmware version 1.0.5.2 in branch [develop](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop), commit [ebe17e5](https://github.com/jnsbyr/esp8266-intexsbh20/commit/ebe17e53d73a85bc9919bbc3bfe0a023fd7addce), that supports setting the disinfection time (see updated [README](https://github.com/jnsbyr/esp8266-intexsbh20/tree/develop#mqtt)).
> 
> The MQTT disinfection command now requires hours (0,3,5,8) instead of on/off.
> 
> Please test and report what happens when you use this new feature.

Hello,

I tried with the new firmware.
I added a selector in my config : 

# Intex Desinfection Time Input Select
  - unique_id: mqtt_intex_desinfection
    command_topic: pool/command/disinfection
    state_topic: pool/disinfection 
    name: intex_desinfection
    options:
        - "0"
        - "3"
        - "5"
        - "8"
        - 
And everything seems to work well for 1 week
Thanks !

SpaWnOL68, 2023-07-08:
> Hello,
>
> I tried with the new firmware. I added a selector in my config :
>
>   Intex Desinfection Time Input Select
>   unique_id: mqtt_intex_desinfection
>   command_topic: pool/command/disinfection [...]

jnsbyr, 2023-08-04:
Thank for testing and reporting back.

FYI: There is a [new release](https://github.com/jnsbyr/esp8266-intexsbh20/releases/1.0.6.2) that includes the features that @SpaWnOL68  has tested and additionally avoids disconnecting the MQTT connection while changing the desired water temperature. The new method, though a little slower than the previous approach, mimics the human behaviour when operating the buttons on the physical Intex panel by "listening" to the buzzer and continuously "reading" the new setpoint value from the display. It may still happen temporarily that the setpoint is changed by more than 1 degree in one step, but if this happens when the desired value has been exceeded the new code will simply reverse the direction so that the desired value is still set and saved.

Background: In issue #15 [RealByron](https://github.com/RealByron) suggested an alternative approach to set the desired water temperature by extending the length of the reply pulse until the falling edge of the latch signal. I tried this variant but had to abandon it, because:

1. extending the pulse length significantly over 2 µs breaks the specification of the Intex original 
2. an edge interrupt might be executed with significant delay while the ESP8266 handles WiFi causing undefined extension of the reply pulse length so that the data line is still blocked by the ESP8266 when the Intex PureSpa is already talking again
3. the code build with ESP SDK 2.7.4 ran into exceptions - the stack trace looked similar to a stack overflow where the newly introduced ISR was called again and again

closed 2023-08-04 by jnsbyr