jnsbyr/esp8266-intexsbh20 #19: Integration with Home Assistant

pfigueira-csm, 2022-07-24:
Hi,
How can I integrate it with Home Assistant?

Clement87, 2022-07-24:
Look here: https://forum.hacf.fr/t/tuto-spa-intex-dans-home-assistant-en-mqtt/11913
It's in French but code is code...

LubosD, 2022-08-10:
Hi, I've started working on an [esphome version](https://github.com/LubosD/esphome-intexsbh20/tree/feature/esphome) of this firmware, but it's currently absolutely untested - it just compiles and flashes. I'm still waiting for my PCB from JLC.

Once the PCB arrives and works, I'll submit a PR.

LubosD, 2022-08-24:
So my esphome branch now works. I haven't yet solved how to nicely report through esphome's C++ APIs that the whirlpool is _not_ connected to the ESP, but usual functions work.

louis49, 2022-09-05:
I'm actually working on a HomeKit version but I will need to change ESP8266 to ESP32.