jnsbyr/esp8266-intexsbh20 #22: Tiny webserver to use instead of mqtt

maddevil84, 2022-11-11:
Would it be possible (even if it's not so secure as you stated in the overview) to add a small webserver / frontend to see and set the values for testing or direct adjustment?
I am not so deep into esp or arduino programming but if somebody could do the first steps I could maybe make the frontend more secure and / or functional.
I guess many people would be interested in having the option to use a small web front-end on this device additionally or instead of the mqtt option.
I really appreciate your work so far.
Thank you and best regards maddevil84

jnsbyr, 2022-11-12:
Sure, its possible to add a webserver to the project. The ESP8266 has ample flash memory for such a purpose and there are a lot of projects out there that show how to do it. Currently I have no intention to add this feature myself due to lack of need and spare time to do so, but I will support anyone who is willing to do job.

jnsbyr, 2023-08-04:
closed due to inactivity

closed 2023-08-04 by jnsbyr

titolini72, 2024-10-28:
Hi maddevil84,
Is you request to have an HTTP Front end service instead of MQTT. Like a web page to control SPA ?

maddevil84, 2024-11-28:
> Hi maddevil84, Is you request to have an HTTP Front end service instead of MQTT. Like a web page to control SPA ?

Hi, the request was to have a small web page  it in addition. Maybe like it is done for "Tasmota". There you have a small and simple web interface  and the possibility to use MQTT.

titolini72, 2024-11-28:
Hi MasDevil84,
Yes this is my understanding. Today I try to pull the new WifiManager with author, based on 8266.
This is a way to setup WiFi and MQTT settings without programmer need. This step is exclusive in the way when WifiManager works, MQTT doesn’t and the other round. 

Your request will run in parallel thus not possible with ESP8266. The solution is to migrate upon ESP32-S3 running at 240Mhz which allows to run both. 

I already built an ESP32-S3 firmware but not yet tested with Intex. 

All the steps have to reach for having this WebServer. This is ongoing. Only a matter of time. 

Regards