jnsbyr/esp8266-intexsbh20 #23: no reaction to commands

swisscryptodev, 2022-11-15:
Hi
I love this project very much, but I don't get it fully working...
Everything works fine except the commands are not executed on the pool. After sending any command, there is no physical reaction of the pool. on MQTT the esp8266 sends an update of the old state.
In the attached example the filter is on and I send the command pool/command/filter off; but the filter stays on and the on the MQTT client we see the state update of the pool with pool/filter on.
Somehow the execution of the pressButton commands has no effect.

[image]

When I press the button on the pool panel, I get the correct state updates on the MQTT.

I have the SB-H20, WeMos D1 mini as ESP8266. Compiler settings are set as requested (160MHz, 4M, IRAM).

Any idea what could be wrong?

jnsbyr, 2022-11-16:
Sending commands to the SB-H20 is really tricky and takes the ESP8266 to its limits, at least in regard to the timing. Receiving the status updates is easy in comparison.

If you have access to a digiscope or a logic analyser you should connect it to the data line and see if the command pulses shown in the readme [diagram ](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/assets/Button-Signalling.png) appear when you send a command. Check before and behind the level shifter FET Q1.

If you do not have this kind of equipment it will not be easy. A mulitmeter will not be much help. Try checking if the FET pins are connected in the right order.

If the command pulses are not generated or generated at the wrong moment nothing happens at the SB-H20.

swisscryptodev, 2022-11-29:
Thanks, Jens for your answer.
I've found the problem. The issue was on the hardware side...: The soldering of the connection BS170, Pin2 to +3.3V was bad!!!
Now it works fine and as expected.
You can label this issue as "solved".
Very nice project! Thank you so much!

jnsbyr, 2022-11-29:
Thanks for your feedback swisscryptodev - enjoy!

closed 2022-11-29 by jnsbyr