jnsbyr/esp8266-intexsbh20 #24: MQTT Server/Broker IP

Telo87, 2023-01-30:
Is ist possible to enter the IP of the MQTT-Server?

jnsbyr, 2023-01-30:
I would assume yes, but documentation of the underlying API [PubSubClient::setServer](https://pubsubclient.knolleary.net/api#setServer) is a little unspecific in this regard, just saying "the address of the server".

This is an overloaded function call so it would be possible to switch to the IP specific function variant if necessary, but it is probably much easier to just give it a try. A web searched regarding this question indicates that it will work (e.g. http://www.steves-internet-guide.com/using-arduino-pubsub-mqtt-client/).

In the config file it would look like this:
`"mqttServer":     "1.2.3.4",`

Telo87, 2023-01-30:
Thanks for the quick response.
I use iobroker with the mqtt broker as server. My other mqtt clients all work, but the intex project just won't work. Unfortunately I have no idea if the problem is with the ESP, the wiring or the software.
The ESP is connected to the WLAN and can also be pinged.

Below are some pictures of my configuration:
config.json
[image]

iobroker Server
[image]

iobroker Client
[image]

Do the data points have to be added manually at iobroker?

Many Thanks

jnsbyr, 2023-01-30:
I have no experience with the iobroker myself. Maybe someone else can provide more specific help. Here are some general ideas:

- check that the ESP is connected to the iobroker on port 1883 - it will try to connect immediately after boot. I would assume that the iobroker has a status display for this. Even if not, your OS should be able to tell you. On Windows and older Linux use "netstat -tn", on newer Linux use "ss". The IP of the ESP and the port 1883 should be shown as "connected". If it is not, there might be something wrong on the ESP side. Connect a TTL/USB adapter to the serial TX of the EPS and listen with a serial monitor (PuTTY, Arduino IDE) at 74880 Baud. If you are lucky you might get some specific error message.
- PubSubClient only supports MQTT version 3.1 and 3.1.1 (see https://github.com/knolleary/pubsubclient)
- without a subscribe the MQTT server will probably not get any data published from the ESP. This is a custom device so the iobroker will probably not be able to do an auto configuration. You might need to add the MQTT topics listed in the [README.md](https://github.com/jnsbyr/esp8266-intexsbh20#mqtt) manually. Test with "pool/model".

closed 2023-02-05 by jnsbyr