jnsbyr/esp8266-intexsbh20 #27: ESP Home installation failed.

RianFlow, 2023-05-28:
Hello. When installing on the wemos d1 mini, I keep getting errors. I've attached the log. maybe you can do something with it. Would appreciate any help.
[logs_esphome-web-819576_run.txt](https://github.com/jnsbyr/esp8266-intexsbh20/files/11583883/logs_esphome-web-819576_run.txt)

jnsbyr, 2023-05-28:
Can't help specifically as I do not have a ESPHome environment.

On first glance it looks like a Makefile/build related problem. It's always a good idea to start at the top:

> src/esphome/components/intexsbh20/SBH20IO.cpp:1:1: error: expected unqualified-id before '.' token
> 1 | ../../../SBH20IO.cpp
> | ^

I don't think that the 1st char on the first source line is the real problem here, as it should be "/*" (start of a comment) in SBH20IO.cpp. Looking closer you see the long relative path for the source file. As I don't have experience with ESPHome, I don't know if this is typical, but I would try to avoid this. What the error really seems to complain about is the "|". This is also not part of the source file, so it must be coming form somewhere else.

All other errors seem to have a similar pattern. So I suggest you check your build environment. Maybe there is someone who knows ESPHome who can offer more support.

RianFlow, 2023-05-29:
ok. Da ich kompletter Neuling auf diesem Gebiet bin, frag ich mich ob ich irgendwas an diesen Dateien ändern muss. An sich sind die Dateien ja leer. Und im Internet finde ich nichts dazu wie man so ein etwas größeres Projekt richtig auf den ESP bekommt. LG

OK. Since I'm a complete novice in this area, I'm wondering if I need to change anything in these files. The files are actually empty. And I can't find anything on the internet about how to properly get such a larger project on the ESP. LG

jnsbyr, 2023-05-29:
ESPHome is a platform of its own, in a way a competitor to Arduino. I assume that it might be possible to make the transition, but it could get hard, because the existing code uses interrupt service routines that must be kept in the RAM and not in the FLASH. These routines use almost all the RAM that is available on the ESP8266 so it might become difficult to make it work, if the platform itself also needs resources of that type. As the code is time critical even at 160 MHz, there is no alternative way to implement it without external hardware. Have a look at [diyscip](https://github.com/YorffoeG/diyscip) if you want to go that way.

If you hope that you might easily expand the current functionality with ESPHome, this might become difficult because of the same reason: The project needs almost all available RAM resources for its own purposes.

My advise: if you "just" want to make it work quickly, stay with Arduino firmware and do the interesting things like a web UI or remote control on the other side of the MQTT server with a smart home application.

RianFlow, 2023-05-29:
Ich habe jetzt versucht das ganze über Arduino zumachen. Hat soweit auch gut geklappt. Nur im Home Assistant finde ich nur den Temperatur zustand und den Device tracker. Muss ich noch irgendwas umbennenen?

I've now tried to close the whole thing via Arduino. Has worked well so far. I can only find the temperature status and the device tracker in the Home Assistant. Do I need to rename anything?

jnsbyr, 2023-05-29:
Wenn die Verbindung zwischen ESP und MQTT Server steht, solltest du in der Lage sein, alle vom ESP gemeldeten Topics zu empfangen (siehe [README/MQTT/Published Topics](https://github.com/jnsbyr/esp8266-intexsbh20#mqtt).

Vermutlich musst du jeden Topic einzeln mit einer Komponente im Home Assistent verknüpfen.

jnsbyr, 2023-08-04:
closed due to inactivity

closed 2023-08-04 by jnsbyr