jnsbyr/esp8266-intexsbh20 #28: Wirering

RianFlow, 2023-05-29:
Hallo. Hat jemand einen genauen Anschlussplan? Mich verwirrt das ich jeden Anschluss 2 mal habe (J1 und J2) An J1 schließe ich die Pumpe an? Und an J2 die Steuerung? LG Florian

Hello.  Does anyone have a detailed wiring diagram?  It confuses me that I have two connections for each connection (J1 and J2) Do I connect the pump to J1?  And at J2 the controller?  LG Florian

Elektroarzt, 2023-05-29:
Ich habe auf der Leiterplatte alle Adern zwei Mal vorgesehen, damit man sich schnell und einfach in den Bus schleifen kann. Wenn man das Kabel zwischen Display und Mainboard auf trennt, müssen die beiden enden ja irgendwie angeschlossen werden, damit das Original Display weiter funktioniert. Die Anschlüsse sind deshalb gebrückt und grundsätzlich gleichwertig. Das heißt, einfach die Aderfarben wie auf der Leiterplatte aufgedruckt anklemmen ... sw = schwarz, ws = weiß usw.

RianFlow, 2023-05-29:
Super. Danke

closed 2023-05-29 by jnsbyr