jnsbyr/esp8266-intexsbh20 #29: Homekit

louis49, 2023-05-31:
Hello! 

I'm working on a HomeKit version but I've some trouble with tempset command that crash HomeKit part : [https://github.com/louis49/esp8266-intexsbh20/tree/homekit](https://github.com/louis49/esp8266-intexsbh20/tree/homekit). If someone can take a look ;-) 
I'm using an adapted version of https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266 that rename #define NO_INLINE because of a conflict with ArduinoJson

Waiting to find a solution, I published a homebridge plugin for interested people. 

You can find it here : [https://www.npmjs.com/package/homebridge-intexsbh20](https://www.npmjs.com/package/homebridge-intexsbh20)

Have a good day

jnsbyr, 2023-05-31:
@louis49 
If I understand the documentation correctly, HomeKit is an alternative to MQTT. It's certainly possible to run it in parallel, but the ESP8266 MCU is not that powerful, so I would opt for a feature selection in *common.h* using a *#define UPLINK_HOMEKIT* (or similar) that excludes the MQTT parts when enabled. This would free up some RAM/FLASH resources and maybe help with your crash. 

As the HomeKit library also requires a 160 MHz this might be a problem. The current code enables and disables WiFi when changing any settings, especially the temperature, see *[SBH20IO.cpp](https://github.com/louis49/esp8266-intexsbh20/blob/homekit/SBH20IO.cpp)*, because otherwise the pulse-timing has too much jitter and the Intex controller will not react as expected. I am not sure that another library that relies on a WiFi connection will be happy about this. The MQTT connection is also interrupted during that time, but re-establishes immediately after the action is completed. It might be necessary to explicitly pause the HomeKit connection during the time critical phase when the WiFi is down or to revive it after the WiFi is back.

A second step regarding clean code may be to implement a new class that handles the required abstraction so that the main code does not need to be changed every time a change for MQTT or HomeKit is needed. Problem here is that the ISRs in *SBH20IO.cpp* must stay in RAM and their size is already at the upper limit.

louis49, 2023-06-01:
Ok, I'll try disabling MQTT, but I'm not far from thinking that it will be essential for me to switch to an ESP32 to complete this project. At worst I have the HomeBridge solution which does the job. It's really a great job this project in any case, it works really very well: congratulations!

jnsbyr, 2023-06-01:
Was thinking in the same direction. The resources and restrictions of the ESP8266 may not be enough to handle both the Intex IO and the HomeKit IO, so switching to the ESP32 may do the trick. You probably know that there are several ESP32 versions around (S1, C3, ...). They all seem more powerful than the ESP8266, but the later versions have a completely different CPU core and behave differently and use a little less energy than the earlier ones. Will need some testing to find the perfect match.

louis49, 2023-06-02:
Finally I managed to make everything work last night: indeed the Wifi disconnection was the cause - there was a bug in the Homekit library that I was using when the Wifi was cut, the connection with HomeKit was cut. I created the necessary branch and added the fix on compatibility with ArduinoJson [https://github.com/louis49/Arduino-HomeKit-ESP8266](https://github.com/louis49/Arduino-HomeKit-ESP8266)

It seems to work in parallel with MQTT without any problem: do you want me to create a feature selection based on #define anyway?

Can we limit ourselves to this correction or do you want me to move forward on the element of CleanCode mentioned to accept my next Pull Request?

jnsbyr, 2023-06-02:
As most users do not need to uplinks, the required uplink should be selectable in a way. With a define the project needs:

- less resources (RAM, flash)
- less dependencies to other libraries, allowing easier build

This may provide a user room for individual extensions and it simplifies the build as the user does not need to download all possible dependencies.

Bringing MQTT and HomeKit support together in one project should be an advantage to the users. The following things need to be done from my point of view:

- Your code needs to be updated to the latest version that also supports the Intext PureSpa SJB-HS model. For this I had to rename the class SBH20IO to PureSpaIO. The new model has 2 additional functions corresponding to 4 new MQTT topics (jet, disinfection). The disinfection time can be selected and read back, but this is not jet supported as it needs a significant rework of the display decoding. This is something I plan to do in the near future.
- Please add the defines to support building for MQTT, HomeKit or both. The defines should also be used to remove all unneeded external dependencies when not active by not including them. Default should be either both MQTT and HomeKit off + #error directive (similar to the new model selection).
- Next I will update the README to describe the support for the new model SJB-HS. When adding the HomeKit support another rewrite is needed and for this your input is required. The aspects to cover are a) feature description, b) build specifics and c) configuration/usage. These aspects could be added to the current README. But it is already getting long, so I am thinking about splitting it in several parts first, e.g. a) general b) hardware c) MQTT d) HomeKit. We can address this after the merge is done.
- For the merging I will create a feature branch "homekit". Once this version looks good to both of us we can merge to the main branch.
- If you agree, I would add you as collaborator to the project after the initial merge, so that there is no need to for me to validate every change you might deem necessary in the future. Only condition here is that you do not make "random" changes to the code and try to respond to issues that are HomeKit related when you have the time.

Elektroarzt, 2023-06-22:
@louis49 
I flashed your HomeKit solution from [GitHub](https://github.com/louis49/esp8266-intexsbh20/tree/homekit) and find it really awesome to have HomeKit functionality in this tiny device! My understanding is to flash your project after installing 'HomeKit-ESP8266' from Mixiaoxiao in Arduino IDE.
The HomeKit devices then appear properly in the HomeKit App but they seem to be not responding regularly. As I have the MQTT path in parallel running and this is responding, I can exclude my network environment as the root cause. Do you experience the same issue or did I do something wrong in building the project?

jnsbyr, 2023-08-04:
closed due to inactivity

closed 2023-08-04 by jnsbyr