jnsbyr/esp8266-intexsbh20 #30: Errors on compiling and compiled code enters a fatal exception loop once loaded

JasonRS1969, 2023-06-01:
During compiling the warnings below are flagged.

Compiling completed and can be uploaded to the WEMO D1, but then enters a continuous crash loop.

/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/MQTTPublisher.cpp: In member function 'void MQTTPublisher::loop()':
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/MQTTPublisher.cpp:132:13: warning: comparison of constant '255' with boolean expression is always true [-Wbool-compare]
  132 |       if (b != SBH20IO::UNDEF::BOOL)
      |           ~~^~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:34,
                 from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/MQTTPublisher.cpp:30:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h: At global scope:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:128:22: warning: 'long unsigned int diff(unsigned int, unsigned int)' defined but not used [-Wunused-function]
  128 | static unsigned long diff(unsigned int newVal, unsigned int oldVal)
      |                      ^~~~

/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/ConfigurationFile.cpp: In member function 'bool ConfigurationFile::load(const char*)':
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/ConfigurationFile.cpp:73:31: warning: catching polymorphic type 'class std::runtime_error' by value [-Wcatch-value=]
   73 |     catch (std::runtime_error re)
      |                               ^~

In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/OTAUpdate.cpp:31:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:128:22: warning: 'long unsigned int diff(unsigned int, unsigned int)' defined but not used [-Wunused-function]
  128 | static unsigned long diff(unsigned int newVal, unsigned int oldVal)
      |                      ^~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:115:22: warning: 'long unsigned int timeDiff(long unsigned int, long unsigned int)' defined but not used [-Wunused-function]
  115 | static unsigned long timeDiff(unsigned long newTime, unsigned long oldTime)
      |                      ^~~~~~~~
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/MQTTClient.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:128:22: warning: 'long unsigned int diff(unsigned int, unsigned int)' defined but not used [-Wunused-function]
  128 | static unsigned long diff(unsigned int newVal, unsigned int oldVal)
      |                      ^~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp: In member function 'void SBH20IO::setup(LANG)':
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:203:66: warning: 'static void SBH20IO::clockRisingISR(void*)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  203 |   attachInterruptArg(digitalPinToInterrupt(PIN::CLOCK), SBH20IO::clockRisingISR, this, RISING);
      |                                                                  ^~~~~~~~~~~~~~
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:281:31: note: declared here
  281 |   static ICACHE_RAM_ATTR void clockRisingISR(void* arg);
      |                               ^~~~~~~~~~~~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:203:94: warning: 'static void SBH20IO::clockRisingISR(void*)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  203 |   attachInterruptArg(digitalPinToInterrupt(PIN::CLOCK), SBH20IO::clockRisingISR, this, RISING);
      |                                                                                              ^
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:281:31: note: declared here
  281 |   static ICACHE_RAM_ATTR void clockRisingISR(void* arg);
      |                               ^~~~~~~~~~~~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp: In static member function 'static void SBH20IO::clockRisingISR(void*)':
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:576:23: warning: 'static void SBH20IO::decodeDisplay()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  576 |         decodeDisplay();
      |                       ^
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:282:38: note: declared here
  282 |   static ICACHE_RAM_ATTR inline void decodeDisplay();
      |                                      ^~~~~~~~~~~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:582:19: warning: 'static void SBH20IO::decodeLED()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  582 |         decodeLED();
      |                   ^
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:283:38: note: declared here
  283 |   static ICACHE_RAM_ATTR inline void decodeLED();
      |                                      ^~~~~~~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:588:22: warning: 'static void SBH20IO::decodeButton()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  588 |         decodeButton();
      |                      ^
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:29:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:284:38: note: declared here
  284 |   static ICACHE_RAM_ATTR inline void decodeButton();
      |                                      ^~~~~~~~~~~~
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/esp8266-intexsbh20.ino: In function 'void setup()':
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/esp8266-intexsbh20.ino:128:31: warning: catching polymorphic type 'class std::runtime_error' by value [-Wcatch-value=]
  128 |     catch (std::runtime_error re)
      |                               ^~
In file included from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.h:34,
                 from /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/esp8266-intexsbh20.ino:53:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h: At global scope:
/Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:128:22: warning: 'long unsigned int diff(unsigned int, unsigned int)' defined but not used [-Wunused-function]
  128 | static unsigned long diff(unsigned int newVal, unsigned int oldVal)
      |                      ^~~~

/Users/jrsouthern/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
/Users/jrsouthern/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);
      |             ~~~^~~~~~~~~~~~~~~~~

JasonRS1969, 2023-06-01:
Adding Detail from Serial Monitor during the crash (this keeps cycling until the power is removed).

-------------- CUT HERE FOR EXCEPTION DECODER ---------------
�@qA|�����PvX	��.�Ѫ�I�"8!�q�T�����DPQQ@��*����nzAl )�~����*1Rt@#A.)5�*���N|��)�~�PZ��~�PZP�UU	�|'�rf cal sector: 1020
freq trace enable 0
rf[112] : 0�SB-H20 WiFi Controller 1.0.2.1
build with Arduino Core for ESP8266 3.1.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffe40 end: 3fffffd0 offset: 0010
3ffffe50:  3ffffeec 00000000 3ffffee8 40211975  
3ffffe60:  00000000 00000000 00000000 00000000  
3ffffe70:  00000000 00000000 00000000 402209d4  
3ffffe80:  3ffffee0 3ffeeef4 3ffe8b0e 402209e5  
3ffffe90:  3fff053c 3fff050c 3fff053c 40220b2c  
3ffffea0:  00000000 00000000 3fff053c 40211980  
3ffffeb0:  40000000 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 402209d4  
3ffffed0:  3ffffef0 3ffeeef4 3ffe8b0e 402209e5  
3ffffee0:  3fff053c 00000001 3fff053c 40220ada  
3ffffef0:  4025294c 40220cf4 00000068 40220180  
3fffff00:  3ffe8b0e 00000001 3ffeee80 40206386  
3fffff10:  4021583c 00000000 000003e8 4020f390  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 3ffeeef4 40210865  
3fffff40:  3ffeefd8 40251a34 3fffff50 4020e3c0 <
3fffff50:  00012480 0000001c 00000000 3ffef164  
3fffff60:  3ffeefd8 00000000 3ffeec58 402088c0  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000000 feef0032 00000000 feefeffe  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3ffef164  
3fffffb0:  3fffdad0 00000000 3ffef138 40211c90  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101771  
<<<stack<<<

jnsbyr, 2023-06-01:
Some of the compiler warning you have posted are known to me, e.g.:

> /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/common.h:128:22: warning: 'long unsigned int diff(unsigned int, unsigned int)' defined but not used [-Wunused-function]

and are fixed in the latest developer version, not jet released. You can still get by checking out from the developer branch.

The warning

> /Users/jrsouthern/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]

is in the PubSubClient library and is harmless in the contextex of this project.

The warning

> /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/SBH20IO.cpp:588:22: warning: 'static void SBH20IO::decodeButton()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]

results from you using a much more recent ESP8266 SDK than the version the project was originally designed for. I consider upgrading to a later version myself to resolve this. On the other hand other users already have reported that an even new SDK can be used and works without changing the code, so this is also harmless in the context of this project.

But the warning

> /Users/jrsouthern/Documents/Arduino/esp8266-intexsbh20/MQTTPublisher.cpp:132:13: warning: comparison of constant '255' with boolean expression is always true [-Wbool-compare]

is new to me and does not make sense if you look at the code. Your compiler does not detect that this is a lambda expression and this will result in undefined behaviour that is probably the cause for the exception.

Please check if all board settings documented at the top of the INO file are correctly set in Arduino.

If the problem persists, please provide more info:

- application version (from common.h)
- Adruino version
- compiler version used during build
- compiler options used during build (included in first output lines during build)

JasonRS1969, 2023-06-02:
Thanks, I'll work through the suggestions above and feedback.

JasonRS1969, 2023-06-02:
R

closed 2023-06-02 by JasonRS1969