jnsbyr/esp8266-intexsbh20 #31: BOM file

thombrink, 2023-06-02:
Can you export the BOM from the Pool SB-H20 WiFi MQTT.fzz file and add to the repo? I don't have fritzing installed (the free version doesn't run at my machine currently).

Elektroarzt, 2023-06-02:
Fritzing is a lab tool from my point of view. I opened fritzing with the project and didn't find any option therefor, too. I don’t use Fritzing usually, as this project was just a test. I'm using KiCAD.
As there are just few components, it is straight forward to read the values out of the [schematic](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/assets/Schematic.png) of jnsbyr. The placement of the components can be read from the silkscreen print on the PCB, so there is no need for a CPL file.

thombrink, 2023-06-02:
Ah I understand. The reason I asked is because I'm a software developer and don't have any low voltage components knowledge. So my idea was to just let jcbplc place the components for me instead, but they asked for those files.

In the meantime I've found someone who's willing to help me sort out the required components and help me place and solder the correctly.

Thanks for the schematics image, I missed that one!

Elektroarzt, 2023-06-02:
Ok, sounds good to jump in the hardware a bit :). Maybe the hurdle for the next steps after having the right components is high, if you do not have the knowledge of sealing the PCB in terms of IP protection and connecting to the bus cable. Please let me know if I can help you with a plug-and-play hardware solution I 3D printed.

[image]

[image]

thombrink, 2023-06-02:
Haha so it seems ;) I do have experience sealing 230v wiring, but your solution looks awesome! If you would like to share the print files for those components, that would be great!

You already saved me a lot of thinking time on what should be the best position of the box :)

Elektroarzt, 2023-06-02:
I thought of publishing the files at printables.com ... this would be the right place I think with a link to this GitHub page. If you like to have a ready assembled unit, mail me at elektroarzt@digital-filestore.de
For publishing the files it takes a bit of work from my side, please give me some time. Otherwise there will be more questions than answers just kicking out the files without comment or any guidance how to use it.

thombrink, 2023-06-02:
No problem! It's a hobby project and it will take a while before the pcb will arrive. However, if I somehow can't get it all working, then I will send you an email!

I'm also thinking about creating a mobile app instead of using HA. Although I haven't decided yet if I will publish it to the store. That depends on whether there is enough interest. Otherwise I will make a separate apk and ipa available.

Thanks sor far!

SpaWnOL68, 2023-06-10:
Hello. Is it possible to have this BOM files? I would like to let JLPCB do the soldering and it requires this files. Thanks

thombrink, 2023-06-10:
Please read the first response of @Elektroarzt he already answered that question I guess.

SpaWnOL68, 2023-06-10:
I read it but i am not really sure about  the exact componant needed for this project.

SpaWnOL68, 2023-06-10:
> I read it but i am not really sure about the exact componant needed for this project. If maybe someone could give more detail I am really interested.

Elektroarzt, 2023-06-11:
I could provide a short BOM with links for e.g. Conrad in Germany. But it always depends on where you live and where you source your components. Be aware ... if you are not familiar with electronics design, it could be really dangerous to modify a pool which is powered by 230V. Remember, you may have children in the pool and in the worst case water everywhere, not only in the pool. If you are not sure, what you are doing, the risk could be too high.
A proposal ... do you have a friend with electronics background who is willing to help mounting everything together? He will know where to source the needed standard components and how to securely modify the pool.

If you anyway insist on the partslist, let me know.

SpaWnOL68, 2023-06-11:
I agree and I don't want to take any risk of course. I know some guy who can do it and I think I am able to follow the Electric schematic. But we cannot buy this project right ? Are you able to help in doing it ? Thanks

Elektroarzt, 2023-06-11:
I could provide the solution you see on the pictures above for self cost. But this is only for SB-H20, as I just have this model. From the MQTT broker you would have to go your own way, because the software is typically very individual, depending on the HA software you use (ioBroker, HASS, Node Red, etc.). The project is not plug-and-play at the moment.
If you like to go that way, leave PN under elektroarzt@digital-filestore.de

SpaWnOL68, 2023-06-11:
Thanks. I have sent you an email.

Elektroarzt, 2023-06-12:
I did not receive your email, sorry

SpaWnOL68, 2023-06-12:
> I did not receive your email, sorry

Maybe in your spam. I have sent you another one.

B4n4n3, 2023-06-15:
Hey @Elektroarzt i have sent you a email on the Weekend too, but i think you dont received it. Maybe Something wrong with your Mail-Server?

thombrink, 2023-06-15:
I don't want to get pushy, but how is the cleanup process going @Elektroarzt? If you haven't had the time to clean up the print file(s) then its no problem to just send/upload them as they are. A friend of mine has a lot of experience with 3d printing and asked me the stl file. I won't bother you if we somehow can't understand the file/drawing.

Thanks in advance!

Elektroarzt, 2023-06-15:
I like to make a short how-to, it's not about cleaning up the models. There will be questions about the contacts to use, how to insert them, the housing and the support that is needed for the print, material and so on. If I kick out the files, I'm sure many people will ask the same questions.
I'm on holiday until Sunday, so next week I will publish the files including sources of the components once WiFi is back in my life.

Elektroarzt, 2023-06-15:
> Maybe Something wrong with your Mail-Server?

Mail slipped in my spam, sorry. Answer goes out.

thombrink, 2023-06-15:
Ah I understand! Enjoy the rest of your holiday and thanks for the work you've done so far!

Elektroarzt, 2023-06-16:
@jnsbyr: I would publish the files and the how-to for the above mechanical solution on printables.com as there is a built-in 3D-Viewer for the uploaded models and it is the main starting point in the 3D printer community. Do you agree? If you have a good idea for linking in to this project, please let me know. Otherwise I would refer to your GitHub page and keep the information there reduced on the mechanics to avoid doubles on both sides. Maybe a link from here to printables.com would then be ideal.

Elektroarzt, 2023-06-20:
The files are published on [Printables.com](https://www.printables.com/de/model/509289-intex-purespa-whirlpool-controller-mechanics)
@jnsbyr: Please check for linking in both directions. If you need something, please let me know.

jnsbyr, 2023-06-21:
@Elektroarzt 
Have rewritten the [Case & Co.](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/README.md#case-cables-and-connectors) paragraph of the README. It now includes a short description of your project and a link.

Please let me know if you have additional info that should be added.

I think this should complete this issue. If you agree, please close it.

Elektroarzt, 2023-06-22:
Fits for me, thanks!

closed 2023-06-22 by Elektroarzt