jnsbyr/esp8266-intexsbh20 #32: arduino flash ESP error

Matze5593, 2023-06-08:
Hello, when I want to flash the ESP D1, I get the following error message:

C:\Users\XxX\Desktop\esp8266-intexsbh20-1.0.3.3\esp8266-intexsbh20\esp8266-intexsbh20.ino: In function 'void setup()':
esp8266-intexsbh20:145:31: error: exception handling disabled, use -fexceptions to enable
     catch (std::runtime_error re)
                               ^
esp8266-intexsbh20:147:22: error: 're' was not declared in this scope
       Serial.println(re.what());
                      ^
exit status 1
exception handling disabled, use -fexceptions to enable

louis49, 2023-06-08:
You need to activate exceptions handling in your Arduino IDE

Matze5593, 2023-06-08:
ok i did it it works.

in the config.json we use the mqtt server "mqttServer": "mqtt.at.home",

set. Is this in the ESP or do you have to enter the IP address of an existing Mosquitto broker here

closed 2023-06-08 by Matze5593