jnsbyr/esp8266-intexsbh20 #35: Wemos d1 mini version

thombrink, 2023-08-11:
Which version of the wemos d1 mini did you use?
I tried using a v4, but it keeps crashing on boot after I upload the code.

Both de normal D1 mini (clone) board and the D1 R2 & mini crash.
[image]

jnsbyr, 2023-08-11:
It should not really matter which D1 mini you use, it can be almost any ESP8266 (see README). To answer you question: The protoype for this project uses a "Wemos D1 R1" in Arduino IDE 1.8.19 and not a "LOLIN(WEMOS) ...".

If you get a crash please check:

- compiler setting described at the top of the INO file
- library versions described at the top of the INO file and in the README
- IDE version described at the top of the INO file and in the README

E.g. it will not work if you use Arduino IDE 2.X. 

If you already checked all this and use the settings and versions that are listed, please post which firmware version you tried to build.

jnsbyr, 2023-08-22:
closed due to inactivity

closed 2023-08-22 by jnsbyr