jnsbyr/esp8266-intexsbh20 #37: Communicate with arduino

fitorie, 2023-08-16:
Dear Sir,

Thanks for sharing your project. It is very interesting.

I expose my issue: 

My solar inverter closes a relay when reaching some photovoltaic sun irradiance parameters.
I want to turn on the Intex spa heater when that relay is closed to derive photovoltaic power excedents to heat the spa.

My problem is that the inverter is 12 meters away from the spa and I don't have wired cable from one to another.
My first idea was to pair 2 arduino using RF 433 Mhz (one in each side) and modify your code in the arduino that controls the ESP8266 in order to turn on heater when the first arduino detects solar inverter relay closed.

However, I wonder if a simpler solution using MQQT or Home Assistant is possible.

Any help is much apreciated.

Thank you very much in advance
Kind regards from Spain

jnsbyr, 2023-08-16:
This is a DIY project for controlling specific PureSpa models. It describes a possible solution (hardware and software) if you have the expertise to manufacture it yourself. It is not a ready to use commercial solution and there is no how-to that describes how to proceed in detail.

I am sure there are commercial relay units available that can be used with Home Assistant and an external heater unit. But if you are talking about the heater in the PureSpa box, then things become dangerous. I can only discourage you from proceeding if you are not an professional electrician or if you do not intend to let a professional electrician plan and execute any modifications. The heater uses mains voltage and even though there is an RCD attached to the PureSpa an unqualified modification can still cost lives.

jnsbyr, 2023-08-22:
closed due to inactivity

closed 2023-08-22 by jnsbyr