jnsbyr/esp8266-intexsbh20 #38: Simplify PCB

fitorie, 2023-08-16:
To simplify soldering PCB, is it possible to use a board level shifter and the capacitor? 
The guy here: https://github.com/RealByron/PicoW-Intex-PureSpa only uses capacitor and commercial level shifter to connect to a Raspberry pico W.
Obviously the board would not be so robust but without termometer, fusible and using commercial board that act as level shifter, things become easier.

I understand this should be possible with Wemos D1

Please confirm if this is possible.

Thanks a million.

jnsbyr, 2023-08-16:
Often there are many ways to achieve a certain goal. But a commercial PCB is not a strict requirement - it just makes things easier to reproduce, once a final design is available.

> ... is it possible to use a board level shifter and the capacitor?

Yes.

jnsbyr, 2023-08-22:
closed due to inactivity

closed 2023-08-22 by jnsbyr