jnsbyr/esp8266-intexsbh20 #39: config file 'config.json' not found

MichelMazzola, 2023-08-17:
Hello,

I tried to modify that project to use it with Domoticz.
I've loaded the project in platformio, compile well be he get the error that the config file 'config.json' not found
when the system start on the console. Have I forget a config key for the ESP8266 plateform?

Thanks in advance for your time helping me.

jnsbyr, 2023-08-18:
Thanks for your feedback.

closed 2023-08-18 by jnsbyr

jnsbyr, 2023-08-19:
@MichelMazzola 

If you want to improve on the subject you should have a look at [PlatformIO Pre & Post Actions](https://docs.platformio.org/en/latest/scripting/actions.html#pre-post-actions).

It should be possible to copy the data folder from the src subfolder to the main folder if it is missing. Please post your solution here if you find a solution.