jnsbyr/esp8266-intexsbh20 #40: Desired temperature regressions

kvj, 2023-08-22:
Hi,

First, thanks a lot the great work on the project!

While playing with the latest release 1.0.8.0, I've discovered few issues:
* Desired temperature field is always empty, it seems like blinking detection code is failing.
* Up/Down buttons aren't being triggered, not by using `setDesiredWaterTempCelsius()` method, not by calling `changeWaterTemp()` directly

I'm having `MODEL_SB_H20` pool.

jnsbyr, 2023-08-22:
I did not perform a runtime test with firmware 1.0.8.0 before, but I tried it a few minutes ago and it is working for me, as did the previous firmware 1.0.7.2.

Please let me know:

- which was the last firmware version you used that worked for you
- what build system you are using (Arduino, vsCode + Arduino CLI, PlatformIO, etc.)
- what library and SDK versions have been used by the build (look for the summary output at the end of the build process)
- does the display start to blink when changing the desired temperature
- do any other commands work (e.g. power on/off)

kvj, 2023-08-23:
> which was the last firmware version you used that worked for you

Not sure I remember did I test the temperature change or not. But at least other basic functions worked.

> what build system you are using (Arduino, vsCode + Arduino CLI, PlatformIO, etc.)

VSCode + Platformio 6.10.0

> what library and SDK versions have been used by the build (look for the summary output at the end of the build process)

```
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
```

I guess here [1] you have a typo, `framework-arduinoespressif8266@3.30102.0` doesn't belong to the `board_build.filesystem` key

[1] https://github.com/jnsbyr/esp8266-intexsbh20/blob/13542c1907572546c26eea5fcef0babc380b40bd/platformio.ini#L20

> does the display start to blink when changing the desired temperature

Nope. Both buttons (Up/Down) aren't being triggered, so also no buzzer.

> do any other commands work (e.g. power on/off)

Other buttons work as expected: Power, Filter, Bubble, Heater. Sometimes from the second attempt, but quite reliable still

jnsbyr, 2023-08-24:
> Not sure I remember did I test the temperature change or not. But at least other basic functions worked.

As I do not have access to your hardware and cannot reproduce the issue with my hardware this is a very crucial aspect to help narrow down a possible cause. Please build and test the previous releases until you find the one that is working for you. But try test described below first.

>framework-arduinoespressif8266@3.30102.0 doesn't belong to the board_build.filesystem key

Thanks, that's a copy&past bug. Should look like this:

```
platform_packages =
   framework-arduinoespressif8266@3.30102.0
```

> Both buttons (Up/Down) aren't being triggered ... Other buttons work as expected

Please increase the int value 380 for PRESS_SHORT_COUNT in PureSpaIO.h:238 to a higher value, e.g. in steps of 10 until it works or until you have reached 500.

kvj, 2023-08-28:
> Please increase the int value 380 for PRESS_SHORT_COUNT in PureSpaIO.h:238 to a higher value, e.g. in steps of 10 until it works or until you have reached 500

I've tried few options over the weekend, and nothing worked:

1. I've increased `PRESS_SHORT_COUNT` few times up until `PRESS_COUNT` value
2. I've also tried to trigger Up/Down buttons using the `set<Whatever>On()` logic, with the respective button values

The desired temperature isn't being detected if I manually press Up/Down buttons. The screen blinks with new value but the variable stays undefined

jnsbyr, 2023-08-28:
Sorry to hear that you made no progress. As I cannot reproduce your issue, please try to find out which was the last firmware version that did work for you.

Vrumfondel, 2023-10-01:
Hi there,
I'm experiencing similar issues like kvj: With  1.0.8.1 controlling is not working at all for me, while with 1.0.7.2 it works quite decent.
With neither of the versions I receive the topics pool/water/tempAct and pool/water/tempSet.

jnsbyr, 2024-05-25:
I just repeated a tests with

- release 1.0.8.1 
- SB-H20 model
- SERIAL_DEBUG disabled
- FORCE_WIFI_SLEEP disabled

I cannot find any issues with my setup. tempAct and tempSet are correct in MQTT and follow any changes on the SB-H20 immediately. Changing tempSet via MQTT also works.

If you haven't already tried: enable FORCE_WIFI_SLEEP with release 1.0.8.1 as it will provide the control behaviour of the previous releases with the disadvantage that MQTT connection gets lost when changing the temperature setpoint and the advantage of a more robust pulse timing. The main problem here is the design of the ESP8266 as the WiFi protocol is handled by the CPU, interrupting the application code anytime. On would need to upgrade to a ESP32S3 to have a CPU core just for the application - not sure if this is worth the necessary code migration.

It might be that the clock speed of the connection between pool mainboard and pool display vary from model to model or by production date or whatever. Significantly different clock speeds will require different pulse timings, independently of the question about ESP8266 and WiFi modem sleep. But with only a single device available to me I have no way to find cause and solution by myself. One option to start would be to measure and compare the display clock frequencies. The README.md specifies the nominal clock frequency as 100 kHz and shows a plot of more than 300 µs -  this can be used as a reference for any comparisons.