jnsbyr/esp8266-intexsbh20 #42: Feature Request: Adapt to ESP32

Vrumfondel, 2023-09-05:
Adapting to an ESP32 environment would have the advantage of a multi-core ECU, so the time-critical interfacing part could run on a seperate ECU, while all the WIFI and UI stuff runns on the other.

jnsbyr, 2023-09-08:
No doubt, but it will:

- increase the costs per board
- probably increase power consumption

Also: the additional CPU power of an ESP32 is not needed for a pure MQTT gateway. Only when you want to run a webserver on the chip things start to look different. But if you have MQTT you will also have a smarthome. So why have another webserver on the chip? But I don't want a discussion about this here.

There are several forks of this project available for more powerful hardware including ESP32 and Raspberry Pi Pico W. Take a look at the closed issues and at https://github.com/jnsbyr/esp8266-intexsbh20/network and you will be able to find them.

A while back I tried to motivate the authors of these forks to help create a new combined firmware version that supports several hardware platforms at the same time, but there was no response. I will not start an this path on my own as this is a fun project for me and it should stay that way. But if there is someone willing to put time into this idea I will support it.

Vrumfondel, 2023-09-09:
Hello,
thanks for the answer.
1. Price increase: Yes, the price for the ESP board would incrase from 1.84€ to 3.97€. Honestly, not a really big jump.

2. Powerconsumption. This is true aswell, the powerconsumption of an ESP32 is higher then the 8266. On the other hand: We are talking about a device which heats  some 100l of water electically, so this is not really counting in my opinion.

Finally I would keep the project anyways flexible for the user to choos the variant, so everyone can choose what he prefers.

I'll have a look to the other projects.

jnsbyr, 2023-09-09:
> Finally I would keep the project anyways flexible for the user to choos the variant, so everyone can choose what he prefers.

No limitations intended from my side. This is a public repo, anyone can fork and mod. 

Please note: When a project has more than a handful of technologically involved users, the project must work reliably for anyone, at least for a defined set of parameters. If you try to support several hardware platforms and IDEs, you, as project maintainer, either know your way around all these and perform regular tests with all these, or you will get quickly swamped with support requests where this or that combination does not work. This typically requires manpower to handle and this project currently does not have the necessary number of volunteers to do more. Have a look [CONTRIBUTING.md](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/CONTRIBUTING.md) if you want to help.

jnsbyr, 2024-05-25:
closed due to inactivity

closed 2024-05-25 by jnsbyr