jnsbyr/esp8266-intexsbh20 #43: Temperature setting less than 20 degrees

algermi, 2023-09-13:
Hello all,

we have now really rocked a lot and I'm just playing with the idea of leaving the hot tub in the winter.  So that the part does not always heats to 20 degrees, would it be possible from your point of view, the minimum temperature to trick? eg to 10 degrees

Elektroarzt, 2023-09-13:
From what I know, the temperature could be tricked by yourself below 20°C. If you want to operate the pool frost safe at e.g. 8°C just leave the pool set at 20°C (hard coded lowest possible value of the Intex board) and shut it off via MQTT once you read 9°C in your IOT software. Then turn it on at 8°C and so on like a simple two-point controller. Remember not to go too closely to <=5°C, as the pool will not heat anymore to avoid damage from pumping freezing water or ice cubes.
Be sure to have a reliable 24/7 connection to your IOT software therefor ...  if the water is freezing in your pump, this would most likely kill it.

jnsbyr, 2024-05-25:
As pointed out by @Elektroarzt: The temperature regulator of the pool will not support values below 20 °C. A possible solution would be to implement a temperature regulator in the ESP8266. 

Pull requests are welcome!