jnsbyr/esp8266-intexsbh20 #45: Cant bring it to work

cyclemat, 2024-05-17:
have tried to compile aover arduino IDE 1.8 and platform IO 
on PIO i bacome an error Tasmota 8266 not avaible 

so ich change the platformio.ini 
[env:d1_mini]
board = d1_mini
board_build.f_cpu = 160000000L
board_build.filesystem = littlefs
build_flags =
	-D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS
	-D VTABLES_IN_IRAM
platform = https://github.com/platformio/platform-espressif8266.git
platform_packages =
   framework-arduinoespressif8266@3.30102.0
monitor_speed = 74880

it build but didnt work no ip adress in my network and no mqtt messages . the  can anybody  help me ?

jnsbyr, 2024-05-18:
More information is required to pin down the problem. 

- your platformio.ini looks good
- check in the logs of your WiFi access point and your DHCP server for any activity of the esp8266
- enable SERIAL_DEBUG in common.h, rebuild, and use the serial monitor at 74880 baud - if the output does not help you to resolve the problem by yourself, post it
- Did you upload the config file to the esp8286? If yes, post your config with the secrets anonymized.
- Which firmware version did you use?

cyclemat, 2024-05-18:
No Activity on my DHCP or on the WiFi No unknown device.
 will try to enable debug in the next days
Yes have Upload the config File.
Upload this after the Weekend. Give IT Problems with space in the ssid ?
Have used the newest Upload with the copiler Error fix.

cyclemat, 2024-05-19:
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00066830
~ld
error: pll_cal exceeds 2ms!!!
rf cal sector: 1020
freq trace enable 0
rf[112] : 0⸮Intex PureSpa SB-H20 MQTT WiFi Controller 1.0.8.1
build with Arduino Core for ESP8266 3.1.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)
config file loaded successfully (has 9 entries)
required entry 'CyCleMat IOT' not found in config file

cyclemat, 2024-05-19:
{
 "wifiSSID":       "CyCleMat IOT",
 "wifiPassphrase": "xxxxxxxxxxxxxx",
 "mqttServer":     "192.168.0.200",
 "mqttPort":       "1883",
 "mqttUser":       "xxxxxxxx",
 "mqttPassword":   "xxxxxxx",
 "mqttRetain":     "YES",
 "firmwareURL":    "",
 "errorLanguage":  "EN"
}

Elektroarzt, 2024-05-19:
Is it possible you modified the source code with your WiFi credentials as well?
The debug log shows "required entry 'CyCleMat IOT' not found in config file", as if the main code searches for this string in the config file!?

cyclemat, 2024-05-19:
my failure had understand that i need to input the config in the common.h sorry im stupid :)

closed 2024-05-19 by cyclemat

Elektroarzt, 2024-05-19:
Just choose your pool type in the common.h and edit the config to your needs and it should work