jnsbyr/esp8266-intexsbh20 #46: Pool topics changes continuously

marcosgar, 2024-05-20:
Hi
First of all thanks for the great job and detailed description.

I have built the PCB following the provided schematics and programmed the ESP8266 using PlatformIO IDE.

Once the board was connected to my SPA, it connected correctly to my WiFi and through MQTT Explorer I was able to see the status topics of the pool. The problem I have is that the topics power, bubble and filter constantly change between on and off and it does not reflect the real state of the SPA. The temperature does show correctly and I have managed to activate the bubbles from my PC after many attempts.

Has this ever happened to you? Do you have any idea what could be the reason why the topics are not received well or that they constantly change?

Thanks in advance

jnsbyr, 2024-05-20:
The pool topics are updated every 10 s, but they should not toggle randomly. Here are some hints how to analyse:

- Rebuild the code with SERIAL_DEBUG enabled in common.h. Maybe add a few extra log outputs, e.g. to MQTTPublisher::loop(), so you can see what values are really provided by the device. Let the debug version run for a short while and watch what happens in your MQTT client and in the serial output at the same time. Note that serial debugging reduces the application stability, so do not be surprised if some things behave even worse (e.g. giving commands) - but it is the easiest way to look deeper into the application.

- Alternatively use something like tcpdump or Wireshark to look into the connection from the device to your MQTT broker. MQTT Explorer is a MQTT client and can only see what the MQTT broker provides. It could be an issue of the MQTT broker setup.

jnsbyr, 2024-05-25:
closed due to inactivity

closed 2024-05-25 by jnsbyr