jnsbyr/esp8266-intexsbh20 #47: Only one way comunication

cyclemat, 2024-05-24:
Have an Problem i have all running but i only become the informations from the Pool temp/on/Off als information are Updated in the Broker but when i send on Off or Set temp Nothing Happens i used this Level shifter
 https://www.ebay.de/itm/152409796871?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=UA5LzFM4QMO&sssrc=4429486&ssuid=0HguCZJXT4u&var=&widget_ver=artemis&media=COPY

Can the Level shifter be the the Problem ?

cyclemat, 2024-05-24:
[image]

Hier das model

Elektroarzt, 2024-05-24:
Could be an issue in the whole signal path. Not easy to say if it's the level shifter. If you have to debug without oscilloscope (I assume you don't have one at hand), it could be straight forward to flash tasmota temporarily as test firmware and toggle the output pins statically on and off via its web interface and measure along the path with a multimeter to see where the signal is lost, if you are not familiar with Arduino IDE and debugging.

cyclemat, 2024-05-25:
Can anybody compile me an working binary so i can Test its Not a Software issue from a libary or Something else ?

Elektroarzt, 2024-05-25:
Try this:
https://tasmota.github.io/install/

jnsbyr, 2024-05-25:
> Can anybody compile me an working binary so i can Test its Not a Software issue from a libary or Something else ?

Building the binary is probably not the cause of your problems, especially if you use PlatformIO to configure your build process.

It could still be a software issue, but of a hardware related kind. Have a look at issue #40. If you think your problem is similar, checkout release 1.0.5.2. If this release works for you, go forward release by release until the problem occurs again and report back if this helped and which releases worked for you.

jnsbyr, 2024-10-26:
closed due to inactivity

closed 2024-10-26 by jnsbyr