jnsbyr/esp8266-intexsbh20 #48: Panel buttons not working with Wi-Fi controller plugged in

Giuse235, 2024-06-27:
Hi, I followed the electrical scheme and the device is working fine and I'm able to control everything via MQTT but whenever the controller is plugged to the pool panel, I'm not able to control the pool using phisical buttons anymore.
I noticed that if I disconnect the DATA pin to the panel, it works again. Can you help me? Thanks

Elektroarzt, 2024-06-27:
Hi ... did you check your level shifter around MOSFET Q1 in terms of pinning and wiring? You could test the function isolated on your workbench if the level is properly shifted between 3.3V and 5V.

Giuse235, 2024-06-27:
I have a doubt that my MOSFET may be reversed (drain and source switched)... May this be the problem? What is the correct wiring for the MOSFET? Can you please tell my how to check it on the workbench? What do you mean with "if the level is properly shifted between 3.3V and 5V"? Thank you

Elektroarzt, 2024-06-27:
Please read this tutorial concerning level shifting utilizing BS170:

https://www.digikey.com/en/blog/logic-level-shifting-basics

You can measure the reverse diode of the MOSFET with a simple multimeter to check the correct orientation of the pins. If it shows around 0.7V, you measure the diode in forward direction.

jnsbyr, 2024-10-26:
closed due to inactivity

closed 2024-10-26 by jnsbyr