jnsbyr/esp8266-intexsbh20 #52: Wifi Web Server

titolini72, 2024-10-25:
Hi all and mainly author 
I spend time to grab Wifi WebServer from Joff. 
I've tested on my side and it works well indeed. However I don't know how to pull this feature toward your repo.

May someone can help me ?
Regards

jnsbyr, 2024-10-26:
This is a duplicate of #22. 

If someone provides a well tested merge request, I will evaluate it. Please make sure that the code you integrate is compatible with the current project license, that the websever is added as feature that can be turned on and off using a define and that the existing project structure remains more or less unchanged.

titolini72, 2024-10-27:
Hi,
Thanks for your quick answer. The WebServer I added has a genuine impact on the code since it replaces the need to upload a config.json (LittleFS) to make it work without an external tool. I will rethink the code to see the feasibility of putting it under feature flag. 
Regards

titolini72, 2024-10-28:
Jens,
After reading #22, just to be clear, the feature aims to avoid the need for a programming tool by enabling Wi-Fi in Access Point Mode to issue SSIDs, MQTT Infos, and so on. This avoid also to re-program ESP whether any change in its env : like new pass phrase, new port, new  user. User doesn't need to disassemble the SPA  for that. 
Hope this is more clear.

jnsbyr, 2024-11-01:
From my point of view a web server is responsible for handling HTTP and/or HTTPS requests, like Apache or Nginx do. But what most users really talk about is a web UI. And a UI can support various features, like setup, status and control. Typically this also needs some "glue" code to combine the UI features with an existing backend, and that "glue" is often called "middleware".

When enhancing the current project these 3 parts (web server, UI and middleware) should be added as separate elements. Additionally the existing code will probably need a few modifications to provide the required events and controls. This should result in a new project structure where the web UI can enabled and disabled with a compiler option.

Form your comments I take it that you will focus on the "setup" feature. Please remember that the ESP8266 platform is EOL. If I would start the project today I would probably use an ESP32-S3. This would also help to avoid problems like #40. 

If you already have a working prototype you should be able to verify if the problem from #40 becomes more apparent when the webserver is actively used. The ESP8266 is just not powerful enough to handle time critical operations and extensive WiFi operations at the same time. The results may vary depending on the number and size of the HTTP objects that are transferred (page, CSS, JS, images, etc.).

titolini72, 2024-11-04:
I agree with your project analysis: this can be handled in 3 chunks. What I did was a setup part. As Setup is in AP mode and the rest is in STA mode, the code overlap is reduced and I don't believe this will have real-time performances. 

As for ESP32-S3, I am sorry I didn't test as I used an old WeMOS-D1 (ESP-12F). I'm pretty sure the code I used for Setup is ready to handle HTTP(S) requests. This is just a matter of time :) 

Nonetheless I keep in mind the other parts. I think for a basic status UI is not that complicate to put in place. I would handle that in other project. I can check on ESP32-R3 at that time if you think both control and HTTP request may have an impact on real time. 

My project (SETUP) is ready for submission if you don't mind.
Best Regards

jnsbyr, 2024-11-10:
Just to be clear, at this time this project has no support for the ESP32 MCU. This would need code migration and testing. As far as I know there are forks of this project that already did this.

From your recent explanation I take it that you originally did not plan any live webserver operations but instead use AP mode for setup. We should keep it limited to that scope and see how it works out. This project has a development branch. You should update develop from main, if necessary, and then create a merge request for you changes.

titolini72, 2024-11-20:
Jen,
I don't know this is me or Read only GitHub but I can't push my change upon you repo. 
`remote: Permission to jnsbyr/esp8266-intexsbh20.git denied to titolini72.`
I take this opportunity to tell you this code works on ESP32-S3 as well. No yet fully tested though, but it boots.
Regards

jnsbyr, 2024-11-22:
You cannot push directly to the repository. For this you would need to be the project owner or a registered project collaborator. Please create a merge request instead. There are several ways to do this. If you need more help, let me know.

jnsbyr, 2024-11-23:
One more thing, as the web server is no minor code change:

Before you create the merge request, please check the [contributing(https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/CONTRIBUTING.md) rules and the [license](https://github.com/jnsbyr/esp8266-intexsbh20/blob/master/LICENSE.txt) of this project. 

If you have created all code and resources you plan to commit yourself, things are easy regarding the licensing. You only need to agree that the code will be made available under Apache 2.0 license.

But if you include code and/or resources that have been created by others, you have to check for license compatibility. As this project uses Apache 2.0, you cannot include anything that uses e.g. GPL. Remember that you have full legal responsibility for this and if you commit to this project you confirm that your contribution is compatible with this project. 

If you have licensing conflicts, please resolve them by replacing the code and/or resources affected. 

Even if there are no licensing conflicts the [licensing & credits](https://github.com/jnsbyr/esp8266-intexsbh20?tab=readme-ov-file#licenses-and-credits) part oft the README must be updated, so please provide the required information (project names, URLs, license names), e.g. in the merge comment, so that I can do a review and make the the appropriate changes in the README.

titolini72, 2024-11-25:
I tried to clean my work as much as possible. In term of license, as most of the code relies on what Geoffroy Hubert did, this won't be an issue.
Nonetheless I failed to create pull request as I'm not a collaborator : this is what interface said BTW.
Would you mind to provide me some help ?

stemac93, 2024-12-01:
I never did this but I think that you have to fork the project and pull on your repo. After you can make the merge request.

jnsbyr, 2024-12-02:
can confirm this workflow works:
- fork the project you want to enhance into your own repository
- start a new feature branch based on the most recent commit from main/master, in your case could be named "setup-webserver"
- commit your changes to your new feature branch
- create a merge request to the branch that is named in contributing description, in this case "develop"

Elektroarzt, 2025-04-04:
Hi @titolini72, just wanted to try your setup-web server branch with no success. This is what I did:
- downloaded your repo branch named setup-webserver
- opened project in Arduino 1.8.18
- #define MODEL_SB_H20 in common.h
- #define WIFI_MANAGER in common.h
- compiled and uploaded sketch
- uploaded littlefs

The project folder looks ike this:

[image]

I'm getting the following error in the serial monitor after resetting the ESP8266:
```
Intex PureSpa SB-H20 MQTT WiFi Controller 1.0.8.1
build with Arduino Core for ESP8266 3.1.2
based on Espressif NONOS SDK 2.2.2-dev(38a443e)
config file loaded successfully (has 10 entries)
Can't read Home page from LittleFS
Home file 'home.htm' not found
```

In your repo the home.htm file is not placed in the data folder of LittleFS, but in html folder. Manually copying the home.htm to the data folder and upload via LittleFS does not work.

Do you have a hint for me?

github-lx2020, 2025-04-04:
Hi,
I am not titolimi72 but I faced the same issue some days ago.
You have to use 7zip to zip the home.htm file into the following file:
home.htm.gz.
Use gzip as compression format.
Then place it in the data folder.
I read that this is a common way for webpages hosted on esp due performance
topics.

Elektroarzt, 2025-04-04:
Works ... thanks a lot 👋

titolini72, 2025-04-04:
Hi Guys!!! 
Glad to hear some people are taking care of what i did. 
Then … I am sorry… I totally forgot to submit the gzip version of html file.  Actually this is not mandatory but having gzip version you win in bandwidth. 
What you did is right. You have to gzip HTML home page then move the file to be populate with LittleFs. I should have documented a little bit more. Sorry guys for having wasted you time. 

Don’t hesitate for further question. 

As for now I work for a complete HTTP server getting rid of MQTT protocol. But this is going to happen only with ESP32-S3. I already have a prototype with S3 with MQTT but need to develop HTTP server. 
I have some concern to switch over Zephyr to handle this project. 

Cheers

Elektroarzt, 2025-04-06:
Thanks for your work! Do you plan to merge your code into the repo of jnsbyr? This enhancement is great, but the original is 2 steps ahead.

titolini72, 2025-04-06:
Well I’m not the owner. This is up to him to merge the change. 
I don’t why this is not already done though.