jnsbyr/esp8266-intexsbh20 #53: Tested on PureSpa SSP from 2015 some functions doesn't work

stemac93, 2024-12-01:
Hi guys I implemented this solution on an Intel PureSpa SSP
All works except:
-Bubble doesn't start (the pool make sound of error)
-Filter when turned on do nothing when turned off acts like the button to lower the temperature
-Temperature set works only going up wrt the temp actually set

The rest works as expected

Elektroarzt, 2024-12-01:
Did you use SB-H20 or SJB-HS setting?

stemac93, 2024-12-01:
SJB-HS

Elektroarzt, 2024-12-01:
If you manage to get the communication  up and running, please share your findings.

stemac93, 2024-12-01:
I don't have experience in the reading of the intex bus communication. If you can give me some material to study I will appreciate. I'm note very expert in programming I'm more hardware oriented in electronic but I think I can understand. Thanks

Elektroarzt, 2024-12-02:
I didn't sniff the communication myself, but jnsbyr wrote about his setup in the  readme, where you can connect with a logic analyzer to the Intex bus and sniff the commands. An analyzer is dirt cheap meanwhile, search for "logic analyzer 24mhz 8 channel" on Ali and you will find the suitable model for around 6€. For the analyzer software [Sigrok PulseView](https://sigrok.org/wiki/Downloads) is a good starting point, as it is open source and supports this analyzer out of the box.

Be aware, that the second part would be implementing the commands into the code. Maybe jnsbyr or any other contributor could help out. I'm also more hard- than software related.

@jnsbyr: a second thought ... wouldn't it be possible to let the ESP8266 sniff the commands and serial it out or let it simply learn the commands, as it is a bus member yet?

jnsbyr, 2024-12-02:
@Elektroarzt: Your idea of "learning" might work, as some button actions already seem to work for @stemac93.

Had a quick look at the code. For starters one could just log isrState.frameValue in PureSpaIO::decodeButton(), e.g.

`DEBUG_MSG(" >%x", isrState.frameValue);`

either by adding this at the top of the method or by temporarily removing all other code in this method. This could also be wrapped with a define to avoid changing the code.

Then start serial debugging, press a button, note the hex code and edit PureSpaIO::FRAME_BUTTON.

@stemac93: give it a try and let us know what happens

stemac93, 2024-12-04:
Perfect, I will do it. Now the pool is in my vacation house and the mqtt broker is connected to my vpn so I can control from there (200km). When I return there (I think not next weekend but the other) I will make all the tests.

stemac93, 2025-03-03:
Hi guys I forked the project and I added the configuration for the PureSpa SSP, now it works perfectly. I sent a merge request.

jnsbyr, 2025-03-03:
Nice! 👍 I will take care of the README.md.

jnsbyr, 2025-03-03:
@stemac93 : 

What is the exact model name that you tested? SSP-H-20-1? SSP-H-20-1C? Or something else (see type shield and manual)? Can you  speculate if your solution will apply to all SSP models?

stemac93, 2025-03-04:
SSP-H-20-1