jnsbyr/esp8266-intexsbh20 #56: Color Code for Data/Clock/Latch Wires ?

CaptnRamius, 2025-04-19:
First of all, thanks to everybody who has contributed to this great project, it sounds fantastic !
I own an Intex PureSpa SB-H20 and I have implemented all the modifications (exactly trying to copy Jens' hardware solution from the ) but I am having trouble obtaining good values from the tapped pool control unit.

My first problem is that I am not 100% sure which of the wires are the data, the clock and which one is the latch (the ESP starts up, so I think I have identified 5V and ground correctly ;-) ).
**Does anybody have the color code for the SB-H20 cable that connects the control unit to the pump/heater ?**

Funnily, yesterday, when I started soldering things, Jens' link for 3D printing (main project page) to the Intex PureSpa connector on thingiverse had still worked, and I remember that CAD drawing showed some color codes. But that link is not working any longer (I guess it has expired today after 6 months?): [Intex PureSpa connectors](https://www.thingiverse.com/thing:4130911)

I have set up a small Mosquitto server to receive messages. In the best configuration so far I am receiving all of the messages below within a few seconds, which seems to be way too many:

> Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/state', ... (7 bytes))
offline
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/ip', ... (13 bytes))
192.168.xx.yy
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/version', ... (7 bytes))
1.0.8.2
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'pool/model', ... (20 bytes))
Intex PureSpa SB-H20
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/water/tempAct', ... (2 bytes))
36
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/water/tempSet', ... (3 bytes))
-99
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'wifi/state', ... (6 bytes))
online
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/error', ... (0 bytes))
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'wifi/state', ... (6 bytes))
online
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
off
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
on
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
standby
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
off

This goes on and on and runs quite quickly and I believe there is a confusion of wires somewhere in my setup.

Of course, I might have some error in the electronics implementation as well. I tried to spot it, but this is quite hard, so I want to negate that there is confusion of the input wires as a first step.

I would be glad if somebody could post the colors !

Happy Easter !

PS: As a newbie question: I see that I can enable debug messages via serial debug but I dont fully understand how this is meant. I guess that the esp has to be connected to a laptop then via USB (like during flashing). However, when connected to the pool at the same time, the 5V from the pool might confilct with the voltage from my laptop's USB port. Yes I know this is also 5V, but there might be small currents flowing from here to there nonetheless, no? Did you guys connect pool and laptop at the same time to the esp for debugging ? Seems way more comfortable for debugging than sending everything through mqqt...

stevengoossensB, 2025-04-22:
Had the same issue... Just found this back in the closed issues:

https://user-images.githubusercontent.com/69749534/121392224-7ce65000-c94f-11eb-8313-5aa678858860.png

CaptnRamius, 2025-04-25:
Thanks ! This looks exactly like what I was looking for !

CaptnRamius, 2025-04-26:
Hi again,
I have double-checked that all wires are connected as proposed. However, I still get "random" messages (reporting switching on and off for various states quickly back and forth). See the protocol above. It seems "in principle" something is sensed, but it is not interpreted in the right way, or not at the right speed.

I am using a D1 mini (AZ-delivery) at 160MHz.
I have no means to debug on the electrical signal level (like oscilloscope) besides measurung simple things like DC voltages, resistance, connectedness.
The next thing I could try is to enable debug messages, but - as stated in the previous post - I was hesitating to connect the D1 mini to the USB port of my computer while it is power through the pool at the same time. And I am not sure this would bring any useful information.
Anybody has a suggestion how to debug this ?
Thanks !

CaptnRamius, 2025-04-26:
Ok, I was searching again through the older closed issue and I found the recommendation to switch on SERIAL_DEBUG.
So I did that and received the following output, maybe somebody sees something ?

> Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/state', ... (7 bytes))
2025-04-26T12:01:33+0200 wifi/state 7 'offline'
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/ip', ... (13 bytes))
2025-04-26T12:01:33+0200 wifi/ip 13 '192.168.29.69'
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'wifi/version', ... (7 bytes))
2025-04-26T12:01:33+0200 wifi/version 7 '1.0.8.2'
Client mosq_sub1 received PUBLISH (d0, q0, r1, m0, 'pool/model', ... (20 bytes))
2025-04-26T12:01:33+0200 pool/model 20 'Intex PureSpa SB-H20'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (3 bytes))
2025-04-26T12:01:33+0200 pool/power 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
2025-04-26T12:01:33+0200 pool/heater 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:33+0200 pool/telegram/led 5 '28674'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:01:34+0200 pool/filter 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (2 bytes))
2025-04-26T12:01:34+0200 pool/heater 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:34+0200 pool/telegram/led 5 '16512'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (2 bytes))
2025-04-26T12:01:35+0200 pool/bubble 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
2025-04-26T12:01:35+0200 pool/heater 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:35+0200 pool/telegram/led 5 '17408'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (3 bytes))
2025-04-26T12:01:35+0200 pool/bubble 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:35+0200 pool/telegram/led 5 '49168'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
2025-04-26T12:01:36+0200 pool/filter 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:36+0200 pool/telegram/led 5 '20482'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:01:36+0200 pool/filter 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:37+0200 pool/telegram/led 5 '49168'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
2025-04-26T12:01:37+0200 pool/filter 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (2 bytes))
2025-04-26T12:01:38+0200 pool/power 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
2025-04-26T12:01:38+0200 pool/heater 7 'standby'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:38+0200 pool/telegram/led 5 '21249'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (2 bytes))
2025-04-26T12:01:38+0200 pool/bubble 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:01:38+0200 pool/filter 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (3 bytes))
2025-04-26T12:01:39+0200 pool/power 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
2025-04-26T12:01:39+0200 pool/heater 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:39+0200 pool/telegram/led 5 '17408'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/bubble', ... (3 bytes))
2025-04-26T12:01:39+0200 pool/bubble 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (2 bytes))
2025-04-26T12:01:39+0200 pool/filter 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:39+0200 pool/telegram/led 5 '28674'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:01:40+0200 pool/filter 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:01:40+0200 pool/telegram/led 5 '49168'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
[... 346 lines cut ...]
2025-04-26T12:02:17+0200 pool/heater 7 'standby'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:02:17+0200 pool/telegram/led 5 '21249'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (3 bytes))
2025-04-26T12:02:18+0200 pool/power 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (3 bytes))
2025-04-26T12:02:18+0200 pool/heater 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:02:18+0200 pool/telegram/led 5 '28674'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:02:18+0200 pool/filter 3 'off'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:02:18+0200 pool/telegram/led 5 '49168'
[... cut ...]
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/power', ... (2 bytes))
2025-04-26T12:02:20+0200 pool/power 2 'on'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/heater', ... (7 bytes))
2025-04-26T12:02:20+0200 pool/heater 7 'standby'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/telegram/led', ... (5 bytes))
2025-04-26T12:02:20+0200 pool/telegram/led 5 '21249'
Client mosq_sub1 received PUBLISH (d0, q0, r0, m0, 'pool/filter', ... (3 bytes))
2025-04-26T12:02:20+0200 pool/filter 3 'off'

CaptnRamius, 2025-05-03:
I am opening a [new topic](https://github.com/jnsbyr/esp8266-intexsbh20/issues/57), since the colors have been resolved and now my problem sees rather related to the signals.

closed 2025-05-03 by CaptnRamius